openscad/openscad #4320: SVG and PDF export do nothing at all

ygoe, 2022-08-04:
When I export to STL or as PNG image, I get the expected result. I am asked for a file name and that file will be written.

Clicking on the SVG or PDF commands from the export submenu, nothing happens. The menu closes but no file picker opens and no file is written. It's as if the event handlers for these menu commands are simply missing.

Is that supposed to work? I've seen some notes about unfinished stuff in the documentation for SVG, PDF isn't even mentioned there. But other issues about problems with SVG or PDF export suggest that it should at least try to do something.

OpenSCAD version 2021.01 on Windows 10

UBaer21, 2022-08-04:
SVG and PDF are 2D formats - what does the console window show?
`̀ERROR: Current top level object is not a 2D object.`

ygoe, 2022-08-04:
So is PNG. I'd have expected to get the same as in the PNG just as SVG or PDF. I was just curious how that looks. So it's useless for me. Why would I use OpenSCAD for 2D content.

Anyway, yes, that's what the console shows. I don't usually use it so it's hidden. Very irritating if nothing happens at all. A better solution would be a message box so that the error message can reliably be displayed. Or maybe the menu command could be labelled "(2D only)" and disabled if there is no 2D content. This is a usability bug.

UBaer21, 2022-08-04:
PNG is an Image while SVG and PDF (container) (and dxf) are vector formats that uses the 2D objects (polygon square, circle and projection) - This is used e.g. for laser cutter, plotter and cutting plotter.  However the console will always give you warnings, errors and status.  Also if you try to save an object (3mf or stl) without rendering first.  This will also show when using the command line interface.
As you wasn't aware what you are doing, i understand that this confused you.  An additional msg box could help in both cases. Also maybe divide the export format list in 2D and 3D.  If you have 3D and 2D objects in scene you will also  just get a warning when try rendering.  (preview can have both at the same time). 

SCAD (2022) also offers an error log to show these if open 
[image]

t-paul, 2022-08-04:
Closing as the issue is resolved. Suggestions for specific improvements should go into dedicated issues.

I'd vote against popup errors, those are quite annoying too. It might be possible to enable/disable exports based on what data is actually available.

closed 2022-08-04 by t-paul

jordanbrown0, 2022-08-04:
Disabling works but can be mysterious.  (Why is this option disabled?)

I'd say that console messages are an appropriate response to execution problems in the program, while popup messages are appropriate responses to user errors in the GUI.  (Like, for instance, the "current tab has been modified since last render" message.)

t-paul, 2022-08-04:
I think menu items can have tooltips explaining the reason (although I have to check that I'm not mixing that up with some other GUI framework).

ygoe, 2022-08-04:
Well, why should I waste precious screen space for the console if it doesn't help me in any way? If there's a problem with my code, a message tells me to look up the details in the console. Doesn't happen very often. Why doesn't this mysteriously non-functional menu item also tell me that it has left a note for me where I'm not currently looking (and it knows that)?