openscad/openscad #4321: Feature request: simple view port animation without rerender

Parkinbot, 2022-08-06:
with more complex designs and modules that are not cached properly (like sweeps) it can take a long time to record a simple viewport travel, e.g. a rotation, say from $vpr=[90, 0, 90] to $vpr=[90, 0, -90]. So it would make a lot of sense to have viewport animations without the need to have each frame rendered and again.

t-paul, 2022-08-06:
That's not possibie in general. Preview only does the 2D image shown in the viewport so unless there's a special case like top level `render()` or top level operation is something lke a `linear_extrude()` there's no 3D geometry that would remove the need for running the preview display operation again for a changed viewport.

Maybe I'm misunderstanding what you mean, so do you have a specific example for clarification?

One totally different option might be adding support for animating a rendered (= mesh calculated object) but that would also, at least for now, suffer from the fact that this operation drops all color information.

nophead, 2022-08-06:
I think the issue is that you can pan and rotate with the mouse in real
time but with vpr etc you have to recalculate everything. I can't see any
easy way around that because the sweep data could depend on vpr.

Parkinbot, 2022-08-06:
@nophead yes that is excactly what I mean. If  I open the viewport control pane, I can easily rotate and translate the view without rerendering. So my suggestion is to automatize that. E.g. denoting a range for each of the variables, as well as FPS and STEP values. And a dump checkbox of course.

t-paul, 2022-08-06:
I think I know now what you mean. I'm not sure if there's a simple way of really just repeating the GUI display step. For the current logic to work based on the variables, obviously the evaluation of those needs to run. So that basically means a completely different feature for specifying animation not dependent on the script but defining the animation properties externally, like some sort of table based keyframe animation, a bit like CSS animations work, modifying properties of a virtual *camera* instead of $vpX affecting the model space.

Actually I like that idea, but I have no idea how difficult it is to implement. A simple start with fixed transition functions might not be that bad (yes... famous last words disclaimer applies).

Parkinbot, 2022-08-06:
For now I have to do such an animation manually using ctrl+shift+C and saving the clipboard to a file or using "File/Export as image" which definitely would be worth to get a shortcut. Then I alter an input field or click on one of the arrows and advance the view to the next frame. 

I think it can easily be implemented in the fashion of a GUI macro recorder without even touching the OpenSCAD implementation. I would expect it to be added to the Viewport-control pane logic: 
Add some input fields to define the ranges, as well as the FPS and STEP values and a start and pause button. Or better reuse the control aggregate from the Animation pane. 
The logic handling the play button event can e.g. simulate change events to the input fields already there as needed by the range definitions and in a linearized fashion. Each change event of an input field will automatically update the viewport which can be snapped and written to a frame00xyz.png. Sounds quite simple to me. 
A more impactful implementation could directly call the appropriate control handlers and avoid the event queue.

MichaelPFrey, 2022-08-06:
the first thing that springs to my mind is to write a small custom qt program that essentially "remote controls" openscad using qdbus. Qdbus can control the viewport and trigger menu items from the UI.

I currently struggle to think of an elegant way of pragmatically exporting images. But... "keeping it simple" would be a full screen screenshot.

"Export to clip board" also sounds like a very powerful idea.
That could be useful option in the export menu.
Note that they way we have implemented qdbus, any menu item automatically gets available on the qdbus interface. (we literally just iterate trough the menu to get the items for the input drivers.)