openscad/openscad #4322: Missing menu commands "thrown together" and "compile" from documentation

ygoe, 2022-08-07:
My model looks good in preview but a part of the difference operation is ignored in render. That part is a polyhedron. After reading a bit, I assume I've put the faces in the "wrong" order. Whatever "wrong" means. The [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#Mis-ordered_faces) has an article about how to handle that. Unfortunately, both menu commands are missing in my application (version 2021.01 on Windows 10). Maybe it's a translation bug, my UI is in German and I can't do anything about it. But no command from the View menu sounds like "Thrown together". And in no menu can I find anything that sounds like "Compile".

* Is that documentation outdated?
* Can I change my app UI to English so that I understand what the documentation says?
* Can't the app re-order faces to be "correct" on its own?

As a quick help, could somebody please add screenshots of those strange menu commands to the documentation? So I could at least match them by their order.

UBaer21, 2022-08-07:
kombinierte Anzeige F12
[image]
In "Einstellungen" you can change GUI localization to keep it in english.

"Compile" probably means  F5  preview or "Vorschau"

t-paul, 2022-08-07:
Please use the mailing list for further usage questions. This is not the right place for general discussions.

https://openscad.org/community.html

closed 2022-08-07 by t-paul

ygoe, 2022-08-07:
Oh dear, what has "Kombinierte Anzeige" (combined view) to do with "thrown together" (zusammengeworfen)?! Definitely a translation bug. And the two separate "Vorschau" (preview) commands are confusing, too. One has F5 and the other has F9 has hotkey. I always thought they were the same, but obviously they're not. Another translation bug?

After a closer look I also discovered the option to disable localisation of the UI. I was looking for something that reads "German" or "Deutsch" and looks like a language selection but failed with that.

That view indeed helped me identify to "wrong" faces as pictured in the FAQ and by try and error I could reorder things until no face was pink anymore. (In a triangle, there's only two options…) Now the render view looks good, too.

ygoe, 2022-08-07:
@t-paul This issue seems to be about translation bugs. Are they also tracked in the mailing list or IRC? (Both are not a viable option for me.)

UBaer21, 2022-08-07:
there is no translation bug.  Combined means two things into one like thrown together. 
However I agree that F9 preview and F5 preview is confusing.  There are F5 and F6 preview and render which using different software to generate the view and depending on that different view modes can be used (and depending on the scene/single object) - that is a bit messy

t-paul, 2022-08-07:
If you have a specific suggestion for better translations, an issue here would be the correct place.

1) "here's a specific suggestion to change X" -> issue here
2) "I found a bug in Y" -> issue here
3) Open questions like "How do I do Z?" / "Can I do Z?" / "Is there a better way for Z?" -> whatever discussion forum you feel comfortable with, although excluding mail and IRC (and Matrix) seems like a harsh restriction regarding official places. This leaves still at least the web forum connected to the mailing list (which is not very good but lacking other options maybe it's at least something) or reddit or whatever other places exist.

t-paul, 2022-08-07:
@UBaer21 yeah, the display options are a bit messy, so if someone has ideas how to improve/fix that would be nice. However that's one of those (rare?) exception cases where an issue "Menu entries are buggy" is not going to help without an actual suggestion attached.

t-paul, 2022-08-07:
Also for the record, "Zusammengeworfene Ansicht" is not going to be the new german menu entry translation :wink:.

UBaer21, 2022-08-07:
@t-paul  maybe  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_User_Interface#OpenCSG_%28F9%29
F9 can be called  openCSG (memory)   and  looks like  "compile" is now labeled "preview" so we have to change that in the wiki.

nophead, 2022-08-07:
I am a native English speaker and I don't understand how those top four entries are supposed to work.

jordanbrown0, 2022-08-07:
Maybe
(a) combine F9 Preview and F10 Surfaces into one, that shows the most recent preview-or-render,
(b) Make F11 Wireframe warn if you haven't done a CGAL-active render since the last change.  (Ideally, also figure out how to have CGAL involved on every F6 render.)
(c) Rename "Thrown together" to something like "Diagnostic".

I am not sure how "show edges" relates to those or to F5/F6, but it should be considered at the same time.

nophead, 2022-08-07:
Seems like F9 and F12 are opposites in F5 mode and F10 and F11 are opposites in F6 mode but are not ordered in a sensible way.

F9 gains a tick mark when selected. and loses it when the others are selected. All very confusing that they only work in conjunction with F5 and F6, otherwise they just blank the screen. They should force an F5 or F6 if one hasn't been done instead of displaying a blank screen.

t-paul, 2022-08-07:
My interpretation is that only F5 and F6 would be actions that cause display of some sort.
The others are options, either toggle or radio groups that should be separated menu structure wise, or maybe even moved into the viewport or somewhere else.
An example would be Cura Prepare vs. Preview and Preview having Layer View and X-Ray View.

jordanbrown0, 2022-08-08:
Concur that F5 and F6 are the "actions", and F9-F12 plus Ctrl+1 "Show Edges" are view modes.

The problem is that F5 and F6 are *also* modes, and F9-F12+Ctrl1 each only work in one or the other F5/F6 modes.

F9 Preview - only if F5 is most recent, else blank
F10 Surfaces - shows last F6 result (even if changed and previewed since then)
F11 Wireframe - shows last F6 result (even if changed and previewed), only if CGAL was needed.
F12 Thrown Together - only if F5 is most recent, else blank
Ctrl+1 Show Edges - Different between the two.  F5 shows triangles and edges from the primitives, cut away but not modified by difference or intersection; F6 shows all resulting edges but not triangles.  F5+render() shows final edges and triangles.

It seems like F9 and F12 should be accessible only if F5 is most recent, and F10 and F11 should be accessible only if F6 is most recent.  (The F10/F11 behavior of showing the last F6 result even if there have been changes does not seem desirable.)

I don't have a truly solid suggestion for how to present all of the variations, but I know that when I want edges or wireframes I always have to stumble around until I get the combination that I need.