openscad/openscad #4323: A method to enable building and accessing data structures more easily

rjcarlson49, 2022-08-09:
I've used a couple of different data structuring methods is OpenSCAD. The structs sub library in BOSL2 works, but it is somewhat hard to set up and use. The data structures themselves, if declared statically, are very readable, which is valuable.

I have also set up non-associative vectors that form the data structures. The declarations themselves then are not very reader friendly unless you add a lot of comments for clarification. Access can be made readable in at least 2 different ways. One is to define variables that represent the offset to a particular field. Like this:

`
a = 0;
b = 1;
c = 2;

v1 = [11, 22, 33];
echo(v1[b]); // echos 22
`

The second method is to create functions to access the fields of the structure. These work fine and are compact in usage, but the syntax is less intuitive.

`
function a(v) = v[0];
function b(v) = v[1];
function c(v) = v[2];

v1 = [11, 22, 33];
echo(b(v1)); // echos 22
`

My suggestion is to interpret "." as an operator such that v.name is implemented as name(v). It makes the code for field access quite readable and intuitive. 

`
function a(v) = v[0];
function b(v) = v[1];
function c(v) = v[2];

v1 = [11, 22, 33];
echo(v1.b); // echos 22
`

This would be very flexible and allow all sorts other things, like vectorname.sum for example. 

My apologies if this fits badly with current meaning of ".". I could find only .x, .y and .z in the documentation, with no more general meaning. This seems to work fine with those usages. I thought for a second about what happens when v1.a is on the left side of an assignment, but given that variable in OpenSCAD are not really variable, I don't think it should be allowed.

I did find this: [#4294 - dict](https://github.com/openscad/openscad/pull/4294) which alludes to a more general dot notation. It's not yet in the build AFAIK and I could not get a general "." to work in the way implied. The dict function and the dot usage in the that example would work for my needs I think.

jordanbrown0, 2022-08-10:
Ref #3088 .

The "read" and "iterate" sides are there.  What isn't there is a way to create objects from OpenSCAD programs; it is only possible to create them from OpenSCAD internals.

There's no public documentation yet because they aren't in a released version and even in the nightly they are created only by functions (textmetrics, fontmetrics, JSON import) that are considered experimental.

I'm pretty sure that I wrote some documentation on textmetrics and fontmetrics, but right now I don't remember where I put it.

See #3684 for textmetrics and fontmetrics.  See #3891 for JSON import.

Because (in the nightly) you can put a function into a variable, as soon as there is a way to create an object you will presumably be able to put a function into an object.  (But hmm.  Probably need a "self" or "this" mechanism.)

thehans, 2022-08-10:
Overloading "." operator to disguise function calls as accessing member data is a very bad idea that will only lead to confusion and headache.

jordanbrown0, 2022-08-10:
What aspect are you complaining about?  Many OO language allow "obj.method()" as a way to call methods and "obj.member" as a way to access members.

If you are complaining about the possibility of having "obj.member" really run a function, yeah, I mostly agree, though at least a few languages allow that through allowing you to define "getters" and "setters".  I regard that as a serious power tool that may have its place, but is probably more dangerous than it's worth.

thehans, 2022-08-10:
I was responding to the original post's suggestion:

> My suggestion is to interpret "." as an operator such that v.name is implemented as name(v).

t-paul, 2022-08-10:
As @jordanbrown0 already said, the infrastructure for objects and access to that structure via "." dot notation already exists, so adding that function mapping would be impossible without breaking that.
Discussions on making creation of those data structures available to the user in #3087 / #3088.

closed 2022-08-10 by t-paul

nophead, 2022-08-10:
Would it work if a function literal is assigned to the member?

```
obj.func = function(x) x + x;
y = obj.func();
```

t-paul, 2022-08-10:
Having function literals in objects, yes, that's normal behavior. However it's unlikely to have assignment like this (also the current parser can't handle "." notation on left side of the assignment). I think the syntax would more target just creating objects which then are immutable (see example in #3087).

rjcarlson49, 2022-08-12:
I'm surprised at the resistance to having x.y invoke a function Y on object x. I'm used to OO languages like Swift and ObjectiveC and this seems quite natural. Are there dangers? Well, probably, but there seem to be so many good uses of this that it's worth it.

I read through a lot of the discussion in #3087 and #3088, though it's mostly quite old by now. I didn't start with OpenSCAD until 3-4 years ago. I found myself agreeing almost completely with Doug Moen's comments. However, then I went back to Doug original 2015 RFCs. The one on functions and modules I REALLY dislike. I find it weird and non-intuitive. I've always disliked the idea of function variables because they're more confusing than they are worth. I can't even imagine a use for them that makes sense other than a common sort routine where you want to pass it a compare function.

OTOH I REALLY like the Objects proposal and it would fully satisfy my needs stated above. The benefits would be huge IMHO. I hope this gets added sooner rather than later. I'm pretty sure it would change my coding style in scad a lot, a measure of its value. There is one exception. I don't see the need for object customization. It seems like it needlessly complicates the proposal. But then, I don't do projects anywhere near the complexity of what someone like nop head does. A quick thought is that a module returning an object could do anything object customization would do.

Since object literals are already running in hidden code it would sure be nice to see that sooner rather than waiting for the full objects update.

Thank you for your attention! And work!