openscad/openscad #4324: Not Watertight

F14Claude, 2022-08-11:
Following my previous request about stl corrections I found this example.
The result with version 2021.01 is not correct nor watertight (look at the vertical slot on front left).
I noticed however this is corrected with version 2022.06.26.ci11957.
For me this new version should be made official shortly.
I am using for some times without any problem
[P129 X5SA Box II.scad.txt](https://github.com/openscad/openscad/files/9307370/P129.X5SA.Box.II.scad.txt)
.

UBaer21, 2022-08-11:
So there is no issue if it is working with a newer version right?
You can use the[ chat](https://web.libera.chat/?channel=#openscad) to discuss those things
Also  i have seen that some objects have no overlap - this can cause problems. 
see the [tutorial ](https://en.wikibooks.org/wiki/OpenSCAD_Tutorial/Chapter_1#Adding_more_objects_and_translating_objects) about using a small overlap

F14Claude, 2022-08-11:
I added overlap where it may be needed, same story with official current version, strange narrowing for two parts (front and right side).
No problem with Development Snapshot 2022.06.26.ci11957.
Surprised you leave other people encountering same problem as I did without warning.
Hope they'll find out a solution.

t-paul, 2022-08-11:
Closing as the issue is confirmed to be resolved.

closed 2022-08-11 by t-paul

UBaer21, 2022-08-11:
Vertex count will go up when there is no overlap. So the snapshot is creating small voids or at least double vertices instead of errors due to non-manifold geometry.  However how could you warn people about a bug that sit deeper in the CGAL renderer.
So as said your model has non overlapping objects. So not "where it may be needed"  but  overlap where coplanar faces touch should be the design premises.  Doubling the vertices and faces without reason is not a good sign.

F14Claude, 2022-08-11:
Seems to me to be the contrary.
Vertex and triangles counts more than doubled adding overlap.
I did add overlap wherever may exist coplanar surfaces as you can see in attached file.
How come could it be a CGAL problem as I obtain bad results from current version and good ones with dev snapshot?
Should be same version of CGAL.

UBaer21, 2022-08-11:
Your file also contain some import geometry.  In general those issues come up from design flaws that the software try to fix and this can cause problems in CGAL .. the fast CSG feature also using a newer version.

here an example of bad design:
```
union(){
cube(10);
translate([10,0])cube(10);
}
this causes 20 facets while with a tiny overlap only 12 facets. (separated 24 facets)
```

F14Claude, 2022-08-11:
cube(10);
translate([9.999, 0, 0]) {
	cube([10.001, 10, 10]);
}
gives me 24 vertex and 44 triangles
and
cube(10);
translate([10, 0, 0]) {
	cube(10);
}
gives me 12 vertex and 20 triangles.
cube(10);
translate([11, 0, 0]) {
	cube(10);
}
gives 16 vertex and 24 triangles.
All based on Free Online stl repair by FormWare.

UBaer21, 2022-08-11:
[image]
this is a render of your first example.
not sure why you getting 24 and 44.  A cube should have 6 sides 8 vertices and 12 triangular faces.

also from your first code of your design i see coplanar faces on the bottom - normally these will be auto corrected but just as example where these problems can occur 
[image]

btw if you use convexity=5 in the linear_extrude() the preview looks nicer