openscad/openscad #4328: Windows cmake error: `missing: HARFBUZZ_INCLUDE_DIRS`

Paulwhy2, 2022-08-14:
Hi, I am trying to compile openscad to contribute in Windows. I'm following these instructions: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Microsoft_Windows

When running CMAKE:
```
cmake .. -G"MSYS Makefiles" -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release -DEXPERIMENTAL=ON -DSNAPSHOT=ON
```

I get the error:
```
-- Boost: 1.79.0
CMake Error at C:/msys64/mingw64/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find HarfBuzz (missing: HARFBUZZ_INCLUDE_DIRS) (Required is at
  least version "0.9.19")
Call Stack (most recent call first):
  C:/msys64/mingw64/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindHarfBuzz.cmake:48 (find_package_handle_standard_args)
  CMakeLists.txt:278 (find_package)
```

However, I do have Harfbuzz installed with pacman and I verified that I have the file `hb.h`. I think that something is wrong with the cmake file [FindHarfBuzz.cmake](https://github.com/openscad/openscad/blob/ea8950bd3cd7f06da333c15463492e484098e591/cmake/Modules/FindHarfBuzz.cmake) but I'm not familiar with cmake and I wonder if anyone who compiles on windows would be able to more easily debug this.

Thanks!

Paulwhy2, 2022-08-14:
I fixed it. I added the path to the hb.h file to the CMakeOutput.log file, and saved saved it:

//Path to a file.
HARFBUZZ_INCLUDE_DIRS:PATH=C:\msys64\mingw64\include\harfbuzz

jordanbrown0, 2022-08-14:
Unfortunately, fixing the cmake output doesn't fix it for the next person.

When I wrote those instructions (updated from a previous iteration), I did verify them with a clean install, but maybe something has changed.

I've fully updated my primary build environment and cloned a new copy of OpenSCAD, and cmake runs OK, but that's not the same as a fresh install.  I'll do a fresh install on a spare system, but first could you please supply the output from `pacman -Q`?

Paulwhy2, 2022-08-14:
Thanks Jordan,
I still get error messages from Cmake:

-- Checking for one of the modules 'glib-2.0'
CMake Error at C:/msys64/mingw64/share/cmake/Modules/FindPkgConfig.cmake:894 (message):
  None of the required 'glib-2.0' found

and

-- Checking for one of the modules 'glew'
CMake Error at cmake/Modules/FindGLEW.cmake:64 (message):
  GLEW not found

Here is pacman -Q:
$ pacman -Q
base 2020.12-1
bash 5.1.016-1
bash-completion 2.11-2
binutils 2.37-5
bison 3.8.2-3
brotli 1.0.9-3
bsdtar 3.6.1-2
bzip2 1.0.8-3
ca-certificates 20211016-1
cmake 3.23.2-1
coreutils 8.32-2
curl 7.83.1-1
dash 0.5.11.5-2
db 5.3.28-3
file 5.41-4
filesystem 2022.01-4
findutils 4.9.0-3
flex 2.6.4-2
gawk 5.1.1-1
gcc 11.3.0-2
gcc-libs 11.3.0-2
gdbm 1.23-1
getent 2.18.90-3
gettext 0.21-2
git 2.37.2-1
gmp 6.2.1-2
gnupg 2.2.35-1
grep 1~3.0-3
gzip 1.12-1
heimdal 7.7.0-3
heimdal-libs 7.7.0-3
inetutils 1.9.4-3
info 6.8-4
isl 0.25-2
jsoncpp 1.9.5-1
less 590-1
libarchive 3.6.1-2
libargp 20110921-3
libasprintf 0.21-2
libassuan 2.5.5-1
libbz2 1.0.8-3
libcbor 0.9.0-1
libcrypt 2.1-3
libcurl 7.83.1-1
libdb 5.3.28-3
libedit 20210910_3.1-1
libexpat 2.4.8-1
libffi 3.4.2-1
libfido2 1.10.0-1
libgcrypt 1.10.1-4
libgdbm 1.23-1
libgettextpo 0.21-2
libgnutls 3.7.6-1
libgpg-error 1.45-1
libgpgme 1.17.1-1
[... 201 lines cut ...]
perl-MailTools 2.21-1
perl-Net-HTTP 6.22-1
perl-Net-SMTP-SSL 1.04-1
perl-Net-SSLeay 1.90-1
perl-TermReadKey 2.38-2
perl-TimeDate 2.33-1
perl-Try-Tiny 0.31-1
perl-URI 5.10-1
perl-WWW-RobotRules 6.02-2
perl-libwww 6.60-1
pinentry 1.2.0-1
pkgconf 1.8.0-1
rebase 4.5.0-2
sed 4.8-3
tcl 8.6.10-1
tftp-hpa 5.2-4
time 1.9-3
tzcode 2022a-1
util-linux 2.35.2-1
wget 1.21.3-1
which 2.21-4
windows-default-manifest 6.4-1
xz 5.2.5-2
zlib 1.2.12-2
zstd 1.5.2-2

jordanbrown0, 2022-08-15:
Your environment looks considerably older than mine.  (Admittedly, I just updated mine today.)

Mine: base 2022.06-1
Yours: base 2020.12-1

And lots of other differences.

I suggest that you `pacman -Syu`.  I don't know whether you need to do it repeatedly.

Make sure that you have all of the dependencies:
```
$ curl -L https://github.com/openscad/openscad/raw/master/scripts/msys2-install-dependencies.sh | sh 
```

(Or since you already have an OpenSCAD repo, you could just run scripts/msys2-install-dependencies.sh .)

jordanbrown0, 2022-08-15:
For instance:  I don't know whether it's truly relevant, but I have a "glib2" that you don't have.

Paulwhy2, 2022-08-15:
Thanks, Jordan.  I won't be able to work on this project until next
weekend.  I'll try this and respond then.

Paulwhy2, 2022-08-24:
I deleted my build and source directories and reinstalled the dependencies, then cmake ran without errors.  I then did a build and ran Openscad from that build.  Everything worked.  Thank you for your help on this.
I'm not sure why I got those cmake errors the first time, but it must have been my mistake.  Perhaps I made a typo, or did something from the wrong directory.

closed 2022-08-24 by Paulwhy2