openscad/openscad #4331: [Feature Request ] Code readability improvement request

meteozond, 2022-08-19:
The greatest problem is that it is very difficult to find out where is figure you are looking for. 

**Shape icons**
May be it could be a bit easier if at least shape icon near the line added? Coloured or highlighted if it is. The eye sees definitely figure faster than the letters. 

[image]

**Rendered sections**
ultimate dream

[image]

**Code structure**
[image]

**Ligatures**
Near the primitives definitions

t-paul, 2022-08-20:
"Code structure" seems unrelated to the first suggestions, so that needs to be split out or removed from this issue.

"Ligatures" What does that mean in this context?

I'm not sure the GUI components currently support custom shapes in the border area, so there might be quite a number of technical challenges, but I suppose that not an argument against having the feature request.

meteozond, 2022-08-21:
> "Ligatures" What does that mean in this context?

I mean rendering operators as icons, something like this:

[image]

Or even this:

[image]

> I'm not sure the GUI components currently support custom shapes

I understand that all of these things are quite complicated - but just want to start conversation.

MichaelAtOz, 2022-08-23:
I don't support this request, well the icons in the editor aspect. 

OpenSCAD is primarily a textual source code based language.

Especially that last example above without keywords. 
That is an entirely different model, see for example [BlocksCAD](https://www.blockscad3d.com/editor/).

With change #4026 the colour of words/symbols in the editor can have more variety. e.g.
[image]

So the OP request to have shapes stand-out, it would be possible to have a colour scheme where there text shape-word (`cube()` etc) are particularly noticeable (chartreuse, aqua ?).

Also the code layout of the OP is not optimal for identifying sections. 
As discussed recently in the Mailing-list, coding [Style](https://lists.openscad.org/empathy/thread/Q6EDBGGB4RF44HFLENSZA3GVZ7YJ5VIH) is important.
There are many ways to layout code, but if the shapes are important, place them separately, e.g.
[image]
But many people do full structured layout, e.g.
[image]

So I suspect a combination of the new color-scheme mechanisms and appropriate layout could achieve better readability, without overloading the source text with graphics.