openscad/openscad #4332: Right double click sometimes messes up rendering

jordanbrown0, 2022-08-21:
Start with a simple cube(10);.

F5 preview, then right double click outside the model.  It might take a few times, but usually not many.  Click and drag to pan or rotate, and observe the ... interesting ... rendering.

[image]
[image]

It may (or may not) be relevant that Thrown Together shows it in purple.

[image]

jordanbrown0, 2022-08-21:
The problem seems to be caused by the double-click handler changing the GL context and not restoring it.  That newly selected GL context - which is the one used for normal preview - is then used by the right-click object selection magic, and damaged.

There are a number of outstanding questions about exactly why things behave the way that they do.

See the PR to follow for more details.

thehans, 2022-08-21:
I wonder if there is a race condition between rendering the graphics frame and the `MouseSelector` modifying GL states... might need to protect each frame render with a lock.

jordanbrown0, 2022-08-21:
InPhase and I looked at that, because MT issues were our first guess.  (Or at least my first guess.)
There's only one thread involved in the picture, as far as we could tell.  See the PR for a lot more detail.

commit 4110d098 (2022-09-18): Save and restore GL context on double click.  Fixes #4332.
commit 8fa6bfce (2022-09-18): Merge pull request #4333 from jordanbrown0/4332

Save and restore GL context on double click.  Fixes #4332.
closed 2022-09-18 by t-paul