openscad/openscad #4335: Color() command "bleeds" into other objects through the difference() operation

TheAndr0id, 2022-08-22:
If one object has a cut and is colored and that cut extends into another object the color of the first object will bleed into the second object.  For example:
```
color("Red") 
  difference(){
    cube(10);
    translate([-1, -1, -1])
      cube([3,3, 12]);
  }
  
translate([0,0,10])  
  color("Blue") 
    difference(){
      cube(10);
      translate([-1, -1, -1])
        cube([3,3, 12]);
    }

```
[image]

The blue shown mixed into the red is from the cut of the blue cube.  It shouldn't be rendered at all since it's a difference operation on a different item.  (For example, if I removed the cut from the Red cube there wouldn't be a cut into the Red cube from the blue one.
(The overlap is used to avoid z-fighting where the two cube cuts meet)

This is only an artifact in preview since colors are removed on a render.

If all colors are removed (or are the same), this generates a smooth cut of both cubes with no z-fighting or distortions.

jordanbrown0, 2022-08-22:
The previewer hates coincident surfaces, whether they are unions, differences, or intersections - including hating surfaces from one difference that happen to coincide with a surface from elsewhere in the model.

Judicious use of render() can help, as can nudging one surface or another just a little bit.

TheAndr0id, 2022-08-22:
I'm not trying to dispute your comment, but using render() kind of just defeats the purpose of the preview doesn't it?  

I don't use the render() function since (as from the wiki) it "Forces the generation of a mesh even in preview mode."

Wouldn't using a bunch of render() calls (in the absolute worse case) just be the same as just rendering the whole model?  I have read that render() can help in some cases with complex objects since it utilizes the cache, but just scattering it around just seems "wrong" to me. 

As for nudging surfaces, I'm doing this now (mostly using linear_extrudes of 2D shapes cut instead of cutting 3D objects), but this is just causing my code to become more complicated to work around an issue with the color() call.  For instance, hollowing out the bottom of a cube is simple:

```
difference(){
  cube([3,3,2]);
  translate([0.5, 0.5, -1])
    cube(2);
}
```
verses this: (which is identical but respects the color() functionality)
```
union(){
  linear_extrude(height=1)
    difference(){
      square(3);
      translate([0.5, 0.5])
        square(2);
    }
  translate([0, 0, 1])
    cube([3,3,1]);
}
```

Which code would you rather maintain?

The only debug/diagnostic tools we have are color() and echo().  I see them both as up there on the importance scale...

MichaelAtOz, 2022-08-22:
That is the karma trade-off for having fast preview. 
[OpenCSG](https://www.opencsg.org/data/csg_freenix2005_paper.pdf) does its best.
I don't think this can be fixed.

> nudging one surface or another just a **little** bit

Like ` translate([-1, -1, -1/2048]) // note power of two is good here`

MichaelAtOz, 2022-08-22:
As the problem is coincident faces you can remove or minimise those in a few ways

This minimises the overlap in Z, hardly noticeable.
```
e=1/2^10; e2=e*2;
color("Red") 
  difference(){
    cube(10);
    translate([-1, -1, -e])
      cube([3,3, 10+e2]);
  }
  
translate([0,0,10])  
  color("Blue") 
    difference(){
      cube(10);
      translate([-1, -1, -e])
        cube([3,3, 10+e2]);
    }
```

Or make one cube slightly offset to the other
```
e=1/2^10; e2=e*2;
color("Red") 
  difference(){
    cube(10);
    translate([-e, -e, -1])
      cube([2+e2,2+e2, 12]);
  }
  
translate([0,0,10])  
  color("Blue") 
    difference(){
      cube(10);
      translate([-e, -e, -1])
        cube([2,2, 12]);
    }
```

And no, it can't automatically do that, [because](https://reprap.org/wiki/Separate_page_on_the_question).

UBaer21, 2022-08-22:
besides that a minimal overlap shouldn't be a problem - it is still dependent from viewing fov ($vpf) and distance ($vpd) also rotation as orthographic / perspective change the behavior. 
You can use a bigger overlap just for the preview. 
```
 s=$preview?1.1 : 1;
cube( [3,3,2] * s );
```

nophead, 2022-08-22:
>using render() kind of just defeats the purpose of the preview doesn't it?

I do precisely that in my library of parts.  I avoid difference as much as possible but if I do use it I wrap that section in a render(). Since, as in this case, it is usually a simple difference it doesn't take too long and is only done once. Union is free in preview so most of my parts are made of lots of unions of linear extrudes of 2D geometry (that is also very fast).

thehans, 2022-08-22:
Duplicate of #501