openscad/openscad #4336: Feature request: alias-module

kwikius, 2022-08-28:
An id aliasing a module and its arguments
(For simplicity allow alias for builtin module {cube, cylinder etc} only to start with).
The alias-module can use dot member operator to access the arguments.

Consider a simple rectangular plate with holes near the corners.

proposed syntax:
```openscad
/*
 create a rectangular plate with holes near the corners.
*/
plate : cube([100,200,3],center = true);  // module_alias
hole : cylinder(d = 5, h = plate.z + 2);  // module_alias
border = 5; // distance from outside of hole to edge of plate

difference(){
   assert(plate.center);
   plate();  // module_instantiation
   for( x = [-1:2:1]){
      for( y = [-1:2:1]){
         pos = [
             (plate.x - (border + hole.d/2)) * x,
             (plate.y - (border + hole.r)) * y, //hole.r == hole.d/2
             -(plate.z/2 + 1 )
         ];
         translate(pos){
           hole();
         }
      }
   }
}
```
existing syntax
```openscad
/*
 create a rectangular plate with holes near the corners.
*/
plate_size = [100,200,3];
hole_dia = 5;
border  = 5;// distance from outside of hole to edge of plate

difference(){
   cube(plate_size,center=true);
   for( x = [-1:2:1]){
      for( y = [-1:2:1]){
         pos = [
             (plate_size[0]/2 - (border + hole_dia/2)) * x,
             (plate_size[1]/2 - (border + hole_dia/2)) * y,
             -(plate_size[2]/2 +1)
         ];
         translate(pos){
           cylinder(d = hole_dia, h = plate_size[2] + 2);
         }
      }
   }
}
```
syntax tested  with parser.y
https://github.com/openscad/openscad/blob/master/src/core/parser.y
```
module_alias
        : TOK_ID ':' module_id '(' arguments ')'
           {
              /* ... */
           }
        ;
```  
        tested with :
```
@>:  bison -fsyntax-only parser.y 
```

UBaer21, 2022-08-28:
```
size=[100,200,3];
d=5;
module plate(size=size)cube(size,center = true); 
module hole(d=d)cylinder(d = d, h = size.z + 2); 
border = 5; 

difference(){
   plate();
   for( x = [-1,1]){
      for( y = [-1,1]){
         pos = [
             (size.x/2 - (border + d/2)) * x,
             (size.y/2 - (border + d/2)) * y,
             -(size.z/2 + 1 )
         ];
         translate(pos)hole(d);
      }
   }
}
```

kwikius, 2022-08-28:
@UBaer21  Thanks for the alternative forms to write in existing syntax. ( And I updated the example from yours). 

**In larger programs do you agree it would be impractical to use 'd' as the name of a global variable? so you need to use global variables with suitably long monolithic names.**

Encapsulating the builtin module in a user defined module doesn't ( currently) allow access to the member variables  The main point of the feature request is to be able to access the members of the  module, instead of requiring an unrelated global symbol.

For that you need an alias to represent the name of the module together with its arguments.

t-paul, 2022-08-28:
The more general idea for this is outlined in #3087. There's no need for introducing a special alias name which probably would create more namespace issues. The strategy of using normal variables then also enable passing this data structure around as usual with any other variable.

closed 2022-08-28 by t-paul

kwikius, 2022-08-28:
@t-paul would be helpful to see how the  "create a rectangular plate with holes near the corners" looks in the new syntax. Is 

For exampe how do you find the x dimension of the plate in the Openscad2 syntax?

t-paul, 2022-08-28:
There is no final solution for this yet, otherwise this would not sit in discussion / proposal mode, but I'd imagine something like:
```
plate = {
  size = [10, 20];
  center = true;
  cube(size, center = center);
};

echo(plate.size.x); // 10
plate(); // <- that's the part where it's not clear if that's the
         // way it should work, if the geometry is created here or
         // already before when assigning to the variable plate
```

kwikius, 2022-08-28:
> There is no final solution for this yet, otherwise this would not sit in discussion / proposal mode, but I'd imagine something like:
>
> ```
> plate = {
>   size = [10, 20];
>   center = true;
>   cube(size, [...]

with my module alias, it should be simple I think
```openscad
plate = {
  value :  cube([10, 20,30], center = true);
};
echo( plate.value.x); // we can look up the cubes x size without instantiating it
plate.value();    // instantiation .. equivalent to cube([ [10, 20,30], center = true);
```
So 'plate' is just a namespace in the C++ sense?

jordanbrown0, 2022-08-29:
Remember to distinguish between a reference to a module (a piece of executable program) and a reference in some sense to a geometric object (the result of executing a program).

> plate(); // <- that's the part where it's not clear if that's the
         // way it should work, if the geometry is created here or
         // already before when assigning to the variable plate

I think it's clear that merely creating the variable plate and assigning it a { } value should not add geometry to the model.  Half the reason you want to be able to do this sort of thing is to examine or modify the geometry before adding it to the model.

Exactly what the syntax is for adding it to the model, I haven't figured out yet.

I don't think we'd want to mix up the module namespace with the variable namespace by saying
```
obj = { size = [10,10,10]; cube(size); };
obj();
```

Hypothesizing an add(csg) operation, I would think in terms of
```
obj = { size = [10,10,10]; cube(size); };
translate([-obj.size.x/2, -obj.size.y/2, -obj.size.z/2])
    add(obj);
```

And there would be something like
```
robj = render(obj);
```
that would yield robj as a variable of type object, with a schema that describes the rendered object.  Perhaps it would still be a CSG tree, but with a rigid schema that consists only of color and polyhedron nodes and a top-level union.

kwikius, 2022-08-29:
The module_alias is an alias for the uninstantiated/non-geometric module
The module alias works like any other module, but with some restrictions and some extra functionality

```openscad
module my_module() {cube([10,20,30],center =true);}
plate : cube([10,20,30],center =true);
// no output yet

my_module();  // output
translate([50,0,0]){}
  plate();   // output
}
```
The gain with the alias is that you can access the elements since (initially at least) alias is restricted to the builtin modules whose elements are known. 
```openscad
echo(plate.x)
```
the alias could also group associated aliases and numeric elements (so long as they dont clash with builtin module element names) : 
```openscad
plate : 
  cube([100,200,3],center = true),
  hole : cylinder(d = 5, h = plate.z + 2),  // nested alias
  border = 10,   // distance from outside of hole to edge of plate
  pos = [50,50,0]
;  // module_alias

difference(){
   assert(plate.center);
   assert(plate.hole.center==false);
   plate();  // module_instantiation
   for( x = [-1:2:1]){
      for( y = [-1:2:1]){
         pos = [
             (plate.x/2 - (plate.border + plate.hole.d/2)) * x,
             (plate.y/2 - (plate.border + plate.hole.r)) * y, //hole.r == hole.d/2
             -(plate.z/2 + 1 )
         ];
         translate(plate.pos){
             plate.hole();
         }
      }
   }
}
```

kwikius, 2022-08-29:
> The more general idea for this is outlined in #3087. There's no need for introducing a special alias name which probably would create more namespace issues. The strategy of using normal variables then also enable passing this data structure around as usual with any other variable.

 I updated the heading post to the proposed syntax in #3087. This syntax also allows nested module_aliases ("Thingies?") to have their own data. I also included some rules to get things to work with modules in the O.P., and this might provide some solutions that could be applicable to #3087. The basic rule is that only one builtin module definition can be included. In the example that would be ```cube([100,200,3],center = true);```

kwikius, 2022-08-29:
I suspect it is also possible to relax the rule allowing only builtin modules for the instantiable member, but for simplicity maybe best to start by allowing builtin modules only

kwikius, 2022-08-29:
> I don't think we'd want to mix up the module namespace with the variable namespace by saying
> 
> ```
> obj = { size = [10,10,10]; cube(size); };
> obj();

I don' see the objection to the above function call instantiation syntax this since it is exactly the same syntax as is used to instantiate modules in current OpenSCAD.

I would argue you don't need the size member variable, as long as you restrict to allowing only one member single_module_instantiation, so you can access its elements as members of obj ( similar to anonymous struct in C)
obj = { cube([10,10,10]); };

The obj alias now allows access to the cubes members, which is a major win 

assert ( (obj.x == obj.y) && obj.center == false);

jordanbrown0, 2022-08-29:
It's not a syntax concern.  It's a namespace concern.  Today, you can have modules and variables with the same name, and there's no collision because module names and variables names live in different namespaces.

I think that being able to look at the CSG tree is an interesting possibility, but for example purposes I didn't want to go there.  I should note that there's some debate as to whether looking at the CSG tree is useful.

MichaelAtOz, 2022-08-29:
I'm not on board.

`a1 = { built-in(some-sort-of-constant-parameter); some-optional-constants=based-on-expressions; };`

> I would argue you don't need the size member variable, as long as you restrict to allowing only one member single_module_instantiation, so you can access its elements as members of obj ( similar to anonymous struct in C)
> obj = { cube([10,10,10]); };

`ac = { cylinder(50, 5, d2=20); };`

`ac.r2` `ac.d1`?`undef` or auto-calculated? What about parameters for `text()` and it will only get more complicated when multiple statements are allowed, worse with user-defined statements)

Alias's (alii?) seem to be modules with big restrictions, where modules can easily do the job.
Over complicating the language. Syntactic saccharine .

jordanbrown0, 2022-08-29:
@MichaelAtOz I'm not sure exactly which part you're unhappy with.

I agree that having a `{...}` expression be limited to a single geometry statement would be wrong.  It should be an arbitrary block of standard OpenSCAD.

You should then be able to look at the CSG tree produced.  I don't know exactly what that would look like, but start by looking at what Design/Display CSG Tree says.

MichaelAtOz, 2022-08-30:
I don't like the 'alias' concept, where you can access the intrinsic primitive with dot notation that has not been specifically specified. ie
```
aliasCube = { 
    cube([10,20,30]); 
}
echo(aliasCube.z); // where did z come from
```
The parameters of `cube()` are `size=x` or `size=[x,y,z]` and `centre=true/false`, and the various `$fN`'s.

`echo(aliasCube.size.x);` maybe, but digging into parameters just doesn't feel right, and only works if you have one primitive.

As my above example of `cylinder()` with a mix of `r1` `d2`, what is `r2` & `d1` do you convert them from the others or `nul` them and how you handle `text()` parameters, fill in the defaults for all parameters or `nul` them to indicate they were not specified? (as knowing whether something was specified would be useful)

If you do that, basically saying primitives are objects with constants reflecting their parameters (ie intrinsic properties).
Then it is not appropriate to add in random other constants.
e.g. `hole = { cylinder(d = 5, h = plate.z + 2); clearance = 0.1;}; `
what does `hole = { cylinder(d = 5, h = plate.z + 2); d = 0.1;}; ` do?
You'd need `hole.cylinder.d` & `hole.d`, or `hole.self.d`? What if you have more than one cylinder?

From further above
```
plate = {  // module_alias
  cube([100,200,3],center = true); //single_module_instantiation **N.B not instantiated here**
  hole = { cylinder(d = 5, h = plate.z + 2); clearance = 0.1;};  // nested alias
  // is the above plate self reference OK?????
  border = 10;  // distance from outside of hole to edge of plate
  pos = [50,50,0];
};
```
Would need to be
```
plate = {  // module_alias
  base = cube([100,200,3],center = true); //single_module_instantiation **N.B not instantiated here**
  hole = cylinder(d = 5, h = base.size.z + 2); 
  clearance = 0.1;
  border = 10;  // distance from outside of hole to edge of plate
  pos = [50,50,0];
};
echo(plate.hole.d,plate.base.size.x); 
```

I'm wavering here, but options are 
all module objects (built-in & user-defined) have constants of their parameters, and nothing else,
`plate.base.size.z; `
above plus extra constants (how to do scope/conflicts),
```
hole = { cylinder(d = 5, h = plate.z + 2); clearance = 0.1;}; 
```
above, but extras are expressed as extra parameters maybe - solves scope
```
hole=cylinder(d=5,h=plate.base.z+2,clearance=0.1);
echo(hole.d,hole.clearance);
```
objects only have constants specifically defined and you can't access parameters, ie below.

```
aliasCube = { mySize = [10,20,30]; cube(mySize); }
echo(aliasCube.mySize.x);
// no access to cube's attributes
```

So not sure...
Also I think the OP alias concept & @jordanbrown0 CSG.Tree don't gel here.
This issue should be focused, do we have user defined objects/attributes, or CSG.Tree.attributes, or both.

I'm may be babbling...

kwikius, 2022-08-30:
> It's not a syntax concern. It's a namespace concern. Today, you can have modules and variables with the same name, and there's no collision because module names and variables names live in different [...]

I would personally try to avoid naming different entities with the same name. I dont see it is a particularly important feature and I would just take the simpler option. Disallow an alias to have the same name as a module var or function. NOTE. This is entirely backward compatible. Nothing breaks.

The module_alias combines some of the functionality of modules with that of variables.
for example both can be instantiated:
```
my_module() ;
my_alias() ;
```

module_alias alos has some functionality of variables

```
x1 = my_alias.x

x2 = my_array.x;
```
This combined functionality is the fundamental feature of a traditional object, [whose purpose is to combine related stuff together](https://en.wikipedia.org/wiki/Object-oriented_programming) :
"OOP (Object-oriented programming) is a programming paradigm based on the concept of "objects", which can contain data and code: data in the form of fields (often known as attributes or properties), and code, in the form of procedures (often known as methods). "

It would be possible to pick through certain combinations modules,functions variables with the same name in a namespace and deduce for example that only functions return values and so the intent of some invocation is to call the function , while the intent of another is to instantiate the module.

The only concern is backward compatibility, so, as long as old code isnt broken, it is perfectly acceptable to say that when using this feature you are more restricted, you can't have a module or a variable with the same name as an alias in the same namespace. I don't think that is a great issue.

If others feel that it is important to allow different entities to have the same name, then you could choose the approach in C/C++ used for structs v functions. Where there is a name collision the struct name can be disambiguated by preceding it with 'struct' keyword, then the default is to assume it is the name of the function. My choice would be to simply call it an error though.

kwikius, 2022-08-30:
> I don't like the 'alias' concept, where you can access the intrinsic primitive with dot notation that has not been specifically specified. ie
> 
> ```
> aliasCube = { 
>     cube([10,20,30]); 
> }
> echo(aliasCube.z); // where did z come from

The feature is already available for arrays (which I admit I didnt know. Thanks to @UBaer21 for showing that)
```openscad
a = [10,2.3];
z = a.z;  // where did z come
```

MichaelAtOz, 2022-08-30:
Yes, but an alias is not an array, and can have different forms, i=aliasNotaCube.z, I'm no against dot notation, but dot.identifiers materialising without definition. 

To extend the paradigm
> "OOP (Object-oriented programming) is a programming paradigm based on the concept of "objects", which can contain data and code: data in the form of fields (often known as attributes or properties), and code, in the form of procedures (often known as methods). "

And definitions of objects and instantiation if objects, and in OpenSCAD objects methods can in include definitions of Geometry and instantiation of Geometry.

MichaelAtOz, 2022-08-30:
To be clear, `cube()` is not an _array_, the size _attribute_ of an cube could be an _array_ (the specific _attribute_ being its 3 dimensions), or it could be three distinct _attributes_, one form of a specific _parameter_ of a cube is an _array_ (`size=a_3_vector`), the other form is a number (`size=a_number`), an _attribute_ exists when a _parameter_ doesn't necessarily, _attribute_ could be default (like `font=""`, `center=false` or `undef`(?) when a _parameter_ is not specified. The `cube()` _attribute_ is 3 dimensions, the _parameter_ can be positional or named, the name of the _parameter_ is `size`.

This is reflected in the CSG Tree
```
cube();
cube(1);
two=2;
cube(size=two);
three=[1,2,3];
cube(three);
cube(size=three);
cylinder();
cylinder(10,20,30);
cylinder(h=10,20,30);
cylinder(h=10,r=20);
cylinder(h=10,d=30);
cylinder(h=10,d1=30);
cylinder(h=10,r1=20,d2=60);
text("hello");
```
which internally seems to use the base form _parameter_ names
```
cube(size = [1, 1, 1], center = false);
cube(size = [1, 1, 1], center = false);
cube(size = [2, 2, 2], center = false);
cube(size = [1, 2, 3], center = false);
cube(size = [1, 2, 3], center = false);
cylinder($fn = 0, $fa = 12, $fs = 2, h = 1, r1 = 1, r2 = 1, center = false);
cylinder($fn = 0, $fa = 12, $fs = 2, h = 10, r1 = 20, r2 = 30, center = false);
cylinder($fn = 0, $fa = 12, $fs = 2, h = 20, r1 = 30, r2 = 1, center = false);
cylinder($fn = 0, $fa = 12, $fs = 2, h = 10, r1 = 20, r2 = 20, center = false);
cylinder($fn = 0, $fa = 12, $fs = 2, h = 10, r1 = 15, r2 = 15, center = false);
cylinder($fn = 0, $fa = 12, $fs = 2, h = 10, r1 = 15, r2 = 1, center = false);
cylinder($fn = 0, $fa = 12, $fs = 2, h = 10, r1 = 20, r2 = 30, center = false);
text(text = "hello", size = 10, spacing = 1, font = "", direction = "ltr", language = "en", script = "Latn", halign = "left", valign = "baseline", $fn = 0, $fa = 12, $fs = 2);
```

Whereas the CSG products have attributes intrinsic to each instantiated primitive
```
CSG after normalization:
((((((((((((cube2 + cube3) + cube4) + cube5) + cube6) + cylinder7) + cylinder8) + cylinder9) + cylinder10) + cylinder11) + cylinder12) + cylinder13) + text14)

CSG rendering chain:
+cube2
+cube3
+cube4
+cube5
+cube6
+cylinder7
+cylinder8
+cylinder9
+cylinder10
+cylinder11
+cylinder12
+cylinder13
+text14
```

So, I'd say a 'cube' _object_ with dimensions of 10x20x30 would have an _attribute_ defined to be a _parameter_ being an _array_ named "size", referred to as `object.size` which can then be referenced further as `object.size.x`.
You could do
```
eps = 0.01;
c1={ cube([10,20,30]); };
c2 = {cube(c1.size*2); }; // [20,40,60] 
c3= {cube([c1.size.x,c1.size.y,c1.size.z*(1+2*eps)], center=true); }; // [10,20,30.02] to punch a vertical hole
// or
c4 = {cube(c1.size*[1,1,1.01], center=true); } // [10,20,30.3]
```

```
dCylinder = {cylinder(h=10,d=20);} // noting I don't like these single primitive unattributed objects
rod = {cylinder(h=10,r=10);}
// CSG.Tree
// cylinder($fn = 0, $fa = 12, $fs = 2, h = 10, r1 = 10, r2 = 10, center = false);
// cylinder($fn = 0, $fa = 12, $fs = 2, h = 10, r1 = 10, r2 = 10, center = false);
```
`dCylinder.d` or `dCylinder.r` or `dCylinder.r1` and maybe `dCylinder.d2`?? 
Given the syntax auto-converts, I'd suggest all are valid and get calculated appropriately, regardless of how it was specified.

Basically you would need to define the valid parameters for each object primitive type.
If you just consider the basic geometry primitives, easy, but what when you extend to methods.
That is why I suggest objects have constants, use the constants in the primitives,
```
c={
  s=[10,20,30]; 
  cube(size=s); // or cube(s)
  translate([s.x/2,s.y/2,-s.z])
    cylinder(h=s.z*2,d=s.x/2,center=true);
```
`c.s.x` is valid, but `c.size` is not.

kwikius, 2022-08-30:
@MichaelAtOz  Based on the above, overall are you happy with the idea that getting access to the attributes of builtin modules  such as the x,y,z dimensions in case of a cube would be a good thing?
(irrespective you might also like it for user defined modules etc etc etc, irrespective what the attributes names are)
My purpose here is to limit the proposal to the minimum possible functionality without closing the door to  future enhancements

kwikius, 2022-08-30:
Note: I have againedited the OP to reflect that t**_he heart of this feature is just a mechanism to access the elements/members/attributes of builtin modules_**. For that a simple syntax would work as originally proposed without impacting on for example https://github.com/openscad/openscad/pull/3087
In this case 
```
my_cube : cube([10,20,30]); 
```
would just be an alias name to access the elements of a builtin module with a particular set of arguents. no extra attributes or functions or transforms etc allowed. The rest is discussed as enhancements, but might be better added to #3087

t-paul, 2022-08-30:
Without any way to actually use the geometry, this seems too limited and strange. When using the geometry it basically brings all the issues that need to be clarified as #3087 without much extra gain. I think it would be much more valuable moving the #3087 topic forward.

MichaelAtOz, 2022-08-30:
Firstly, continuously editing the OP makes progressive discussion difficult.

> my_cube : cube([10,20,30]);

We are back to that you seem to think the only way to do a cube is `cube([x,y,x]);`
That is just syntactic saccharin.

```
// in a library
function Center(def) = ( is_bool(def[3]) ? def[3] : false );
module Cube(def) cube( [def.x, def.y, def.z], center=Center(def) ); // note caps Cube & Center
//

my_cube = [10,20,30,true]; // add any extra parameters and if you want a named function to extract them
a_default_cube = [1,2,3]; // default is without specifying center
echo( my_cube.x, my_cube.y, my_cube.z, Center(my_cube) ); // possibly some enum type evolution could help the later
// maybe "enum .center = 3;" & thus "my_cube.center;" ??
// or just 
Center = 3; // defined in library
echo( my_cube.x, my_cube.y, my_cube.z, my_cube[Center] );
Cube(my_cube); // instantiate cube.
```
You would need to do so for each primitive, but it is not particularly onerous.

I don't support a point solution for limited benefit which may well interfere with the general object solution.
Even though progression of such seems slow, it is better to have a well considered solution to language evolution.

MichaelAtOz, 2022-08-30:
I do like
```
enum .center = 3; // define dot as vector offset
my_cube.center; // == my_cube[3]
```
Can we do that??

kwikius, 2022-08-30:
> Without any way to actually use the geometry, this seems too limited and strange. When using the geometry it basically brings all the issues that need to be clarified as #3087 without much extra gain.

Not sure what is meant by using the geometry. Could you explain a little more?

kwikius, 2022-08-30:
>  I think it would be much more valuable moving the #3087 topic forward.

 #3087 example has only functions variables as members, whereas this feature fundamentally about accessing the attributes of modules. On thinking more,  I think they are complementary and ( assuming the minimal functionality described in the OP and the ':' syntax)  the feature could be useful  to provide a name to instantiate a module in the thingy in #3087
```
o = {
  my_cube : cube([1,2,3]);
  pos = [3,4,5];
};

translate(0.pos){ o.my_cube();}
```

kwikius, 2022-08-30:
incidentally on testing this out a little by converting some existing openscad code, it certainly seems useful to me. 
I notice now that translations are somewhat wordy which suggests the same syntax could be reused
```
tx : translation([20,30,40]);
x = tx.x;
tx(){my_cube();}
```

jordanbrown0, 2022-08-30:
I should perhaps note that my idea is intended to be a synthesis of the "objects" thread and the "examine geometry" thread.

The objects thread in #3087 and friends wants to have a syntactic way to build what (so far) we've called an object, what some languages call an associative array or dictionary.  So it wants a syntax something like JavaScript's
```
x = { a: 1, b: 2};
```
... but we already have a syntax a lot like what's inside the braces, in regular OpenSCAD:
```
x = { a=1; b=2; };
```

At the same time, the "render as function" discussion in #3956 wants a way to have an expression that evaluates geometry and returns it as data.  The original proposal looks like:
```
data = render() { module(); ... };
```

On the one hand we have { assignments }, and on the other hand we have { geometry }; what would we get if we merged the two into one concept?  It seems like we'd get an object (dictionary/etc) that contains both assignments and geometry.

I worked through some ideas at
https://docs.google.com/document/d/14siqbr9CyGuA2tY6dvTqiF3Hf9zFMyefk7YdaUJSn3Q

One thing that I inserted was the idea that the direct result of this construct would be a CSG tree, rather than being a fully rendered polyhedron.  It seemed wrong to *require* rendering in order to have geometry available as data.

MichaelAtOz, 2022-08-31:
> I'm done commenting. I don't support a

I should note I'm not the honcho of OpenSCAD, but just expressing my experienced opinion FWIW.

@jordanbrown0 Perhaps such discussion should happen elsewhere, I'm not up to speed due to IRL PIT(everything) matters consuming my time, but #3088 presumably is the place, I'll endeavour to read your google doc, but keep in mind this is very meta, be careful with 'geometry' there is the stages of meta-data describing geometry, and the end result, sections or collections of representations of 3D (or2) geometry. In OpenSCAD context, Syntax -> CSG Tree -> vertexes (still just a meta version of space). I had a glance & see you're onto it, but just 'geometry' can be taken to be any of those stages to different readers. Adding 'objects' which could contain any of them just goes [metaphysisical](https://en.wikipedia.org/wiki/Metaphysics).

kwikius, 2022-08-31:
More thought brings me to the conclusion that in general what would be most useful would be a way to make a module first class, so you can tag it with a name, pass it to functions etc.
A simple solution, which also addresses the namespace issue is to reuse the module keyword
```openscad
my_cube = module cube([10,20,30]);
```
You can access the elements of course:
```openscad
x = my_cube.x;
// or 
x1 = module cube([1,2,3]).x ;  // doesnt gain much but for completeness
```
This also leads to being able to access the attributes of a user defined module
```openscad
module  m(a){
   my_cube = module cube(a);
}
h = module m([1,2,3]).my_cube.z;  // the attributes of the module are also available
assert( h == 3);
module m([1,2,3]).my_cube() ;// instantiate cube into CSG tree
````

kwikius, 2022-09-07:
Continuing work on this feature. The .scad sources in https://github.com/kwikius/openscad/tree/WIP-first-class-module/tests/data/scad/experimental/first_class_module mostly now should work and instantiate stuff, baring regressions. I will have to figure out how to get tests working...