openscad/openscad #4341: Feature request: Viewport  into floating window and fullscreen

Parkinbot, 2022-08-30:
As the size of a PNG or clipboard image depends on the physical screen size, I am currently working with floating windows to get a maximum size output. Working with such an organization turns out to be quite odd and impractical. Therefore
- I miss a **fullscreen output**
- I miss a way to **define a viewport size independently of the screen resolution**

Further, the viewport is displayed in the App's main window which also carries menu, symbolbar and statusbar. In order to work with a large viewport (which is inevitable to get output with a serious resolution) all other panes (Editor, Console, Viewport control ...) have to be floating. With this odd design you need to organize all these floating windows, say on a first screen, manually, and find the menu and viewport symbolbar with the viewport in the main window on the second screen. As OpenSCAD stores this design (and in lack of a template View/standard view), you have to reorganize these windows back into a docked design, the next time you start OpenSCAD.
- I miss **view templates**

Wouldn't one expect at least the menu to go with the editor and most of the other panes to be able to dock to the editor? The same holds for the Customizer? This means 
- I miss a **more practical window hierarchy**, where editor as well as all other panes can stay docked in the main window where the menu is and each pane **including the viewport ** can optionally be made floating independently of the others.
 
This would allow a more practical workflow: Start with a docked design and make the viewport floating on demand, instead of having to pull and organize all other panes into floating windows forth and back or resize them to get a larger view.

t-paul, 2022-08-30:
Export independent of window size is on the way via #4261.
Floating main window is difficult with the current GUI code, as it's the center widget. That structure might have been a bad choice in the past but will require some bigger changes to fully restructure the GUI.

I don't see much chance for this to be done as one big topic with it's huge impact, it might be better / more realistic to go with smaller separate changes moving into a new direction. I'll keep it open as I think it's all good suggestions but limitations on both developer time and constraints imposed by the GUI Framework

Parkinbot, 2022-08-30:
I know that touching the odd window organisation is a big mess as so many platforms are involved. However, it was discussed years ago already. And the longer you wait the lower the change gets it will ever get changed, since the code base grows and grows. But maybe it can be scheduled as one of the main goals for a next major release.

t-paul, 2022-08-30:
There is no such thing as scheduling other peoples spare time. We can try motivating people doing specific things, but the bigger the change is the more difficult that gets too. The only possible option would be proposing this as Google Summer of Code project, but that's once a year and depends on a participant actually picking that topic.

Parkinbot, 2022-08-30:
You are right it always depends on the available dev team time. I think OpenSCAD is close to jumping into another league and as a power user of OpenSCAD I feel that such proposals and feedback should not be withheld. They are meant and should serve as improvement not as criticism ...

t-paul, 2022-08-30:
Agreed, and the suggestions certainly get my vote.