openscad/openscad #4342: openscad example "roof.scad" fails ver 2022.08.30

donwade, 2022-08-30:
I've tried building it with -DEXPERIMENTAL=1 and -DEXPERIMENTAL=ON
I've even tried  putting in a "#error ouch" in the source code for built-ins to see if it is being compiled in (it is, the error is thrown on compile).

Simple openscad example "roof.scad" fails.
 
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
ECHO: version = [2022, 8, 30]
WARNING: Experimental builtin module 'roof' is not enabled in file roof.scad, line 23
WARNING: Ignoring unknown module 'roof' in file roof.scad, line 23
WARNING: Experimental builtin module 'roof' is not enabled in file roof.scad, line 27
WARNING: Ignoring unknown module 'roof' in file roof.scad, line 27
WARNING: Experimental builtin module 'roof' is not enabled in file roof.scad, line 32
WARNING: Ignoring unknown module 'roof' in file roof.scad, line 32
WARNING: Experimental builtin module 'roof' is not enabled in file roof.scad, line 36
WARNING: Ignoring unknown module 'roof' in file roof.scad, line 36
WARNING: Experimental builtin module 'roof' is not enabled in file roof.scad, line 40
WARNING: Ignoring unknown module 'roof' in file roof.scad, line 40
Compiling design (CSG Products generation)...
Geometries in cache: 0
Geometry cache size in bytes: 0

Scopeuk, 2022-08-30:
if running openscad from the command line to use experimental features they have to be enabled by passing the enabled argument. see
https://github.com/openscad/openscad/wiki/Experimental-Features#how-does-it-work-from-user-perspective

adding "--enable=roof" to the command should enable this feature. via the ui this would need to be enabled in settings

closed 2022-08-30 by donwade

Scopeuk, 2022-08-30:
Being able to quickly enable or disable features at runtime lets developers compare behaviour. roof perhaps doesn't fall into this category however features like lazy union and fast csg do not modify the scad code but do change performance and the resultant model if used.

UBaer21, 2022-08-30:
And as it is "experimental" it is not enabled by default. So it is available in the download of dev snapshots but not interfering.

jordanbrown0, 2022-08-30:
Requiring the runtime switch means that the user has to explicitly opt in to using an experimental feature that might be removed or changed before its final release.