openscad/openscad #4343: Silent Polyhedron Union Failure with CGAL

JonMcLin, 2022-08-31:
Version: 21.01
Behavior manifests on both Windows 10 and Ubuntu 22.04.

I have a module which produces the union of a skewed five-sided pyramid and a cylinder through the apex of the pyramid.
This is what "preview" shows:
[image]

And this is what I am trying to create.  No errors or warnings are produced.

When I render, this is what I get:
[image]

Note this was part of a larger model, and I created a shell by differencing two different sizes of the above.  In that version, the smallest outside face of pyramid disappeared in the preview, although no error was generated.  Rendering  (in that model) produced a "mesh not closed!" error.   No such error exists in this limited-scope single-instance version.

I set $fa=0.2 and $fs=0.2 thinking I might just need a finer mesh.  It would then never complete rendering.  I had to modify the file in a text editor to remove those settings so I could reopen it in Openscad to produce the above.

Code snippet.  Dimensions are actually from the antique part I'm trying to replicate:

```
function inches(val)=val*25.4;

// Bottom features
drainOD=inches(1/2);
drainID=inches(3/8);
drainCenter=inches(2.875);
drainHeight=inches(0.25);  // actually 3/16
pyramidWidth=inches(3.875);
pyramidLength=inches(8);
pyramidHeight=inches(1/2);

module pyramid(width,length,height, xpt,ypt){
    x=width/2;
    y=length/2;
    z=height;
    points=[
    [x,y,0],[x,-y,0],[-x,-y,0],
    [-x,y,0],[xpt,ypt,z]
    ];
    faces=[
    [0,1,2,3],[0,1,4],
    [0,3,4],[1,2,4],[2,3,4]];
        polyhedron(points,faces,5);

    };
    
   module bottom(width,length,height,xpt,ypt, cylD, cylH){
       union()
 {
           pyramid(width,length,height, xpt,ypt);
  S  translate([xpt,ypt,0])cylinder(h=cylH+height,d=cylD);                  
           };
};
 
bottom(pyramidWidth,
    pyramidLength,
    pyramidHeight,
    0, drainCenter, drainOD,drainHeight);
    

```

jordanbrown0, 2022-08-31:
Use View / Thrown Together to see that the bottom and the small triangle are backwards.

Also:  Try Design / Flush Caches before your F6 render to see if it shows errors.  Unfortunately, erroneous objects get cached too, and so their errors may not show again.  (or something like that).

[image]

[image]

Here's a fixed face list:
```
    faces=[
    [3,2,1,0],[0,1,4],
    [3,0,4],[1,2,4],[2,3,4]];
```

JonMcLin, 2022-08-31:
Thank you.  That fixed the outside, although now my difference() no longer works properly.  More to troubleshoot.

Obviously, I missed a layer of order dependency on the polyhedrons.

closed 2022-08-31 by JonMcLin

jordanbrown0, 2022-08-31:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#polyhedron
has some advice on the rules for polyhedra and how to build and debug them.