openscad/openscad #4344: Possible parallel build issue for 2021.01 with qmake build

waebbl, 2022-08-31:
I received an issue with parallel builds for 2021.01, when using the default qmake based build[1]. The issue seems somewhat random and is not reproducible in a stable manner. It manifest in the file parser.hxx not being found by lexer.l.

```
src/lexer.l:36:10: fatal error: parser.hxx: No such file or directory
   36 | #include "parser.hxx"
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:6119: objects/objects/lexer.o] Error 1
make: *** Waiting for unfinished jobs....
```

Any ideas?

[1] https://bugs.gentoo.org/856430

t-paul, 2022-08-31:
The qmake based build is not supported anymore. The CI builds used to run `make -j2` and did not show that issue so no tips or ideas unfortunately.

closed 2022-08-31 by t-paul

waebbl, 2022-08-31:
I'm wondering now. In https://github.com/openscad/openscad/issues/3665#issuecomment-778783461, dated 14 Feb 2021 you told me, that cmake is not yet the officially supported build system. Version 2021.01, for which this bug is, was released one week before you wrote this. So this version should still use qmake as the official build system.

If you won't support qmake any longer, you should IMO **urgently** release an official version that uses cmake so distributions can update their packages and not rely on a not yet officially supported cmake build system or use git HEAD for their packages.

> The CI builds used to run `make -j2` and did not show that issue so no tips or ideas unfortunately.

I'm aware of this and wrote, that it doesn't have stable reproducability.