openscad/openscad #4347: Last two snapshot builds seg fault when using the Round-Anything library

yeroca, 2022-09-07:
I'm not exactly sure what is causing this seg fault, but it happens only on the two most recent snapshot builds even the 2022.09.05 build.   It seems to only occur when using the Round-Anything library.  I'm seeing the same problem on Windows 10, and on the AppImage build for Linux (running Fedora 36).

Note: the attached .txt file is a .scad file, but github won't let me attach it as a .scad

[Round-Anything.zip](https://github.com/openscad/openscad/files/9522054/Round-Anything.zip)
[antgazebo.txt](https://github.com/openscad/openscad/files/9522062/antgazebo.txt)

yeroca, 2023-01-10:
Still crashing in OpenSCAD-2023.01.04

kintel, 2024-04-01:
Works for me on macOS as of 2024-03-24
Any updates for other platforms?

t-paul, 2024-04-01:
No crash with with both Manifold and CGAL using Debian package (OBS Build) 2024.03.27.nightly (git f2d4c1b) and AppImage 2024.03.28.ai18952 (git f2d4c1b62).

With Manifold: `WARNING: [manifold] Minkowski hard-crashed, falling back to Nef operation.`

With CGAL + Preview: `ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.`

With CGAL + Render only: No warning

kintel, 2024-12-22:
Closing as it no longer crashes

closed 2024-12-22 by kintel