openscad/openscad #4349: Feature Request : Module literal

kwikius, 2022-09-09:
To complement #3077  Modules should also have a mechanism to become first class , which means that we can tag a module and its arguments with a name, pass it to a function , return it from a function, store it in array etc, without instantiating it into the CSG tree. 

Note that this is not the same as the Object Literals proposal #3087, since we already know exactly what a [module](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/User-Defined_Functions_and_Modules#Modules)  is , whereas so far we don't have an clear definition of what an "Object" is or might be in OpenSCAD 

The ModuleLiteral feature has an implementation at https://github.com/kwikius/openscad/tree/WIP-first-class-module . This feature now runs and outputs graphical objects at the viewport and I would suggest the best way to get to know the feature would be to download it and build it and try running some examples, since that will give a better idea of how it works than abstract discussion.

To get you started look in the examples ModuleLiterals directory https://github.com/kwikius/openscad/tree/WIP-first-class-module/examples/ModuleLiterals/

There is some documentation at https://github.com/kwikius/openscad/blob/WIP-first-class-module/doc/ModuleLiteral/moduleLiteral.md

The feature is a pure extension and is entirely backward compatible with existing openSCAD code

Note also this feature request does not address member access. Though that would be nice, the basic ModuleLiteral mechanism does not require it

jordanbrown0, 2022-09-09:
Two notes:

* "Member access" is a whole different story.  You can't talk about members until you start to execute the module.
* Objects are not completely undefined.  The development snapshot has had read-only data-only (no geometry) objects for a while, for the textmetrics and import-json features.

jordanbrown0, 2022-09-09:
What you've got in parser.y is a great deal more than you've described in moduleLiteral.md.

I think you need to back off to a more or less straightforward analogy to function literals, which would (syntactically) probably be nothing more than
```
expr
...
    | TOK_MODULE '(' parameters ')' '{' inner_input '}'
```

kwikius, 2022-09-09:
> What you've got in parser.y is a great deal more than you've described in moduleLiteral.md.
>
> I think you need to back off to a more or less straightforward analogy to function literals, which would [...]
 Apart from the member access part 'module_alias_tail' ,  which parts of the current syntax  aren't useful?

kwikius, 2022-09-10:
Currently in the proposal the following 2 syntaxes are equivalent 
```scad
square = module { translate([-5,-5,0]) cube([10,10,10]);},
square = { translate([-5,-5,0]) cube([10,10,10]);},
```
The second syntax could instead be used for an object, if it is decided that an object is not the same as a module. It doesnt bother me that much personally
(If second syntax used for "object" presumably putting module_instantiations in it would be an error?)

Meanwhile it is easier to remove than add syntax so will leave as is

kwikius, 2022-09-12:
Closing this. I would like to work on it, but not much time at the moment. Will try a pull requets if I get anything useful done

kwikius, 2022-09-25:
Reopening this. I hope to do a pull request at some point, but got various tidying up to do.
I have updated the names in the source according to https://github.com/openscad/openscad/pull/3087#issuecomment-1244103228

I have removed the module ref '.' member lookup from the grammar and the code, for a separate PR. That would make this much more useful, but it is still useful without it, and it keeps things simpler.

I have also removed the  ```v = {...};``` syntax, so now  it is required to do ```v = module{...};```
Apart from anything else I think it makes the intent clearer.

The bug affecting module reference module_instantiation in for loops has been solved, so now[ module_references work correctly in for loops](https://github.com/kwikius/openscad/blob/WIP-first-class-module/examples/ModuleLiterals/foreach4.scad).

Alos now module_references are [equality_comparable](https://en.cppreference.com/w/cpp/concepts/equality_comparable). two module_references will compare equal if one is initialised from another. This makes it possible to [use module_literals as keys in a map for example](https://github.com/kwikius/openscad/blob/WIP-first-class-module/examples/ModuleLiterals/info2.scad#L12).

Also updated the examples
https://github.com/kwikius/openscad/tree/WIP-first-class-module/examples/ModuleLiterals
Typical example:
```openscad
/*
Instantiate some extrusions in various poses using ModuleLiteral feature - simplification 2
*/
use <MCAD/regular_shapes.scad>
use <MCAD/boxes.scad>

extrusionInfo = [
   [[0,0,0],[0,0,0],module(height) { 
      translate([-3.5,-3.5,0]) cube([7,7,height]);
   }],
   [[10,10,0],[22.5,5,0], module(height) cylinder(d = 7, h = height, $fn = 30)],
   [[20,20,5],[45,10,0], module(height) triangle_prism(height,5)],
   [[30,30,10],[67.5,15,0], module(height) octagon_prism(height,4)],
   [[40,40,15],[90,20,0], module(height) {
      translate([-3.5,-3.5,0]) roundedCube([7,7, height],2,false,false);
   }]
];

for(elem = extrusionInfo)
   translate(elem[0])
      rotate(elem[1]){
         height = 40;
         m = elem[2];
         m(height);
      }
```

https://github.com/kwikius/openscad/blob/WIP-first-class-module/examples/ModuleLiterals/foreach4.scad

kwikius, 2022-09-29:
OP updated to reflect new more finished branch https://github.com/kwikius/openscad/tree/module_literal_v2.

closed 2022-12-29 by kwikius