openscad/openscad #4350: fast-cfg problem with MeshTool

dizcza, 2022-09-10:
Originally opened in https://github.com/fieldOfView/Cura-MeshTools/issues/28. Please refer to the issue description there.

The guys at Cura-MeshTools don't seem responsive. But I'm using their tool in Cura, which forces me not to use the `fast-cfg` optimization flags in OpenSCAD. Even if I come up with a fix on their side, I'm not sure if my work will be regarded in a PR merge. Could you have a look from your side? There is something fishy going on with the `fast-cfg` flag enabled that makes MeshTools panic.

UBaer21, 2022-09-10:
You can use lazy union to create multi object part at top level. So you just don't need to split.  Cura-MeshTools shows some weird behavior which has nothing to do with openSCAD.

dizcza, 2022-09-10:
> You can use lazy union to create multi object part at top level. So you just don't need to split.

I don't quite understand how to use this feature. I've enabled it, flushed cache, rerendered, saved to an stl file but still it's opened as a single object in Cura. Splitting required.

UBaer21, 2022-09-10:
it will not union() at top level. So a loop or 
```
cube(10);
sphere(5);
```
will be saved as two objects not one.
Objects within an explicit union(){} will be just one Object.
Also try using 3mf format.

dizcza, 2022-09-10:
You also need to add that

1. Objects within a module() are merged in one object even with the lazy-union feature.
2. Only works for 3mf format.

Unfortunately, I like putting everything inside `module`s to keep the code neat (to be able to render different objects using one file and not clash the global namespace). So your suggestion does not fit my needs. Thanks for the feature though, I didn't know it.

You may close the issue if you find there is nothing to do with it.

UBaer21, 2022-09-10:
There might be future developments to allow a better export of multi parts (e.g. for multi material print). Lazy-union was for faster render with the side effect to create multiple objects in 3mf files.  However having multiple parts you want to keep separated within one module seems to be a design flaw. I would suggest to use an switch/option variable for that module so you can call it to access the single part.
```
Part(option=1);
Part(option=2);
```

Then fast-CSG is still experimental and so there may be also changes. If there are errors in the exported file this would be an issue or bug, but as far as i know is that the files are met the standards. And Meshtools is not up to.  You can try the  "sort-stl" feature maybe that makes a difference how Meshtools handling the file. Via command line option you can test with ASCII and BINARY stl too - if that changes anything. (none should make a difference but maybe it fixes your workflow)

dizcza, 2022-09-10:
> However having multiple parts you want to keep separated within one module seems to be a design flaw

This is because I'm treating a *module* as a function in a typical programming language. And functions in OpenSCAD are very limited in what they can do. In particular, OpenSCAD's function becomes convoluted if I start making local variables there. It seems not to be designed to have a large body. Therefore, I'm misusing the concept of a module to achieve my needs.

I see there is work in progress making OpenSCAD OOP (v2) but, to be honest, I was surprised that 7-years old code written in OpenSCAD runs just as fine as a week old... not many programming languages can claim such behavior. Because the API of OpenSCAD is simple and limited, has always been.

closed 2022-09-10 by dizcza

UBaer21, 2022-09-10:
"typical" is a very subjective term. But you are right that SCAD is a functional language. The functions are very powerful though, when you know how to use them and often this means to have recursive elements. 
Maybe have a look at [OpenSCAD Functional Programming Utilities](https://github.com/thehans/funcutils) or [Functional Open SCAD](https://github.com/thehans/FunctionalOpenSCAD)

jordanbrown0, 2022-09-11:
> However having multiple parts you want to keep separated within one module seems to be a design flaw. 

Not if the intent is to design for multiple materials.  We can't do that today, but we'd like to be able to.  You should be able to have one module invocation create a multi-colored object.

UBaer21, 2022-09-11:
@jordanbrown0  Yes because we don't have that feature - you call it a design flaw.  If i design a car without wheels it is a design flaw until cars can fly - pretty simple. So if the intent is to have a module for multi materials you need a discriminator may be a variable or color or some group(id) ..  if you put objects into a union or module without - it is a design flaw. If the objects don't touch you can separate the hulls later with other software but if they touch like they would for multi-material you can't.

jordanbrown0, 2022-09-11:
[ I would take this to a private conversation, but I don't have your e-mail address and github doesn't seem to have DMs. ]

Sure, but lazy-union is not a solution to multi-material printing if you can't have a module create a multi-material object.  My immediate reaction is that lazy union is not a solution to any *modeling* problem.  It might be a helpful optimization, but does not let you make more interesting models, except in trivial cases.