openscad/openscad #4353: "ERROR: Parsing aborted by unknown exception" after enabling fast-csg

thehans, 2022-09-14:
Running openscad 2022.09.05.nightly on Linux Mint 21.

When previewing the following script (which has some tweaks to resolution made by me), I get the message in Console:
"ERROR: Parsing aborted by unknown exception"

I am pretty sure the wording of this error is a red herring, as I assume this occurs during geometry evaluation, and not "parsing".

```OpenSCAD
// Fillet function
// from https://github.com/clothbot/ClothBotCreations/blob/master/utilities/fillet.scad

$fa = 1;
$fs = 0.25;

module fillet(r=1.0,steps=3,include=true) {
  if(include) for (k=[0:$children-1]) {
	children(k);
  }
  for (i=[0:$children-2] ) {
    for(j=[i+1:$children-1] ) {
	fillet_two(r=r,steps=steps) {
	  children(i);
	  children(j);
	  intersection() {
		children(i);
		children(j);
	  }
	}
    }
  }
}

module fillet_two(r=1.0,steps=3) {
  for(step=[1:steps]) {
	hull() {
	  render() intersection() {
		children(0);
		offset_3d(r=r*step/steps) children(2);
	  }
	  render() intersection() {
		children(1);
		offset_3d(r=r*(steps-step+1)/steps) children(2);
	  }
	}
  }
}

module offset_3d(r=1.0) {
  for(k=[0:$children-1]) minkowski() {
	children(k);
	sphere(r=r);
  }
}

r = 2;
steps = ceil(2*PI*r/4/$fs); // rough estimated fragments for a 90 degree fillet
fillet(r=r,steps=steps) {
	cylinder(r=5,h=10);
	cube([10,10,2]);
	rotate([30,30,30]) cylinder(r=1.0,h=50,center=true);
}
```

Here is a view of which experimental features I have enabled:
(Disabling the topmost "fast-csg" feature allows the script to preview without error)
[image]

rcolyer, 2022-09-15:
With #4354 this is revealed to be "ERROR: Compilation aborted by exception: Unauthorized intersections of constraints"

A perusal of the CGAL source finds this explanatory comment for the exception condition in Constrained_triangulation_2.h:
```
    // if the segment (or a subpart of the segment) that we are trying to constraint is already
    // present in the triangulation and is already marked as constrained,
    // then this is an intersection
```

Additionally, a single run and only one run, I got a different behavior and received this error on the console:
```
WARNING: [fast-csg] Corefinement corefinement mesh intersection #45 failed with an error: CGAL ERROR: assertion violation!
Expr: it_poly_hedge!=edge_to_hedge.end()
File: /usr/include/CGAL/Polygon_mesh_processing/internal/Corefinement/Visitor.h
Line: 1278

WARNING: [fast-csg] Corefinement corefinement mesh intersection #67 failed with an error: CGAL ERROR: assertion violation!
Expr: it_poly_hedge!=edge_to_hedge.end()
File: /usr/include/CGAL/Polygon_mesh_processing/internal/Corefinement/Visitor.h
Line: 1278
```

thehans, 2022-09-16:
Updated issue description after #4354

Also, I just noticed that when this exception happens, the progress bar gets sort of "orphaned", it stops moving, but nothing is being processed.  Then if you try F5 again you get a **second progress bar**, and so on.

kintel, 2024-11-28:
Closing as fast-csg has been deprecated

closed 2024-11-28 by kintel