openscad/openscad #4356: Additional detail on `include` vs `use`

jordanbrown0, 2022-09-16:
One of my correspondents was asking about `use` and `include`.

I wrote:
> "include" is absolutely straightforward.  When file A includes file B, it is exactly as if B was dropped at that point in A.  That's simple, but means that absolutely everything in B is visible to A.
>
> "use" is more interesting.  When A "uses" B, a number of things happen:
>
> *    A can see B's modules and functions.
> *    A cannot see B's global variables.
> *    B cannot see A's global variables.
> *    B cannot see A's modules and functions.
> *    B's top-level module invocations are not executed.
>
> These help you to keep A and B separate.  B can have variables that configure its operation, without interfering with A's variables.

He found that presentation useful and said it should be in the manual.  Looking over the [existing manual page](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Include_Statement) I think that it adds the parts about not being able to see each others' variables, and about B not being able to see A's modules and functions.  Also a concise bullet-point list of behaviors may be helpful.  I missed the part about being able to override values supplied by an include file.

I'm happy to do the editorial work, but not tonight so I thought I'd record this in an issue.

rcolyer, 2022-09-16:
I like that way of expressing it, and think it would help bring clarity.  Just also, "absolutely everything in B is visible to A and everything in A becomes visible to B."  I made use of this text-substitution consequence in pre-function-literal days for https://github.com/rcolyer/plot-function/blob/master/plot_function.scad
```
// Note:  For use as a library this file must be included with "include",
// not "use", so that it can access the functions defined in the including
// scad file.
```

hsuenaga, 2022-10-03:
Please add a note about dynamic scoped $variables if you will update the manual.

It seems module included with "use" statement has some special context. Here is an example. case1 has identical result, but other cases don't.

example module:
```
$test=10;

module A()
{
	echo($test=$test);
}
```

case 1-use:
```
use <testM.scad>

A($test=1);
A($test=2);
A($test=3);
```
case 1-use result:
```
ECHO: $test = 1
ECHO: $test = 2
ECHO: $test = 3
```
case1-include:
```
include <testM.scad>

A($test=1);
A($test=2);
A($test=3);
```
case1-include result:
```
ECHO: $test = 1
ECHO: $test = 2
ECHO: $test = 3
```
case2-use:
```
use <testM.scad>

let($test=1) A();
let($test=2) A();
let($test=3) A();
```
case2-use result:
```
ECHO: $test = 10
ECHO: $test = 10
ECHO: $test = 10
```
case2-include:
```
include <testM.scad>

let($test=1) A();
let($test=2) A();
let($test=3) A();
```
case2-include result:
```
ECHO: $test = 1
ECHO: $test = 2
ECHO: $test = 3
```
case3-use:
```
use <testM.scad>

$test=1;
A();
```
case3-use result:
```
ECHO: $test = 10
```
case3-include:
```
include <testM.scad>

$test=1;
A();
```
case3-include result:
```
ECHO: $test = 1
```

OpenSCAD version:
```
OpenSCAD version 2021.01
```

jordanbrown0, 2025-03-08:
Behavior of $ variables at the top level of use<> files is discussed (though not really resolved) in #3881 and #3781.

jordanbrown0, 2025-03-08:
I took a stab at this:  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Include_Statement

closed 2025-03-08 by jordanbrown0