openscad/openscad #4357: Clicking the mouse on the preview screen makes a weird preview and rendering

dizcza, 2022-09-18:
Open the `logo.scad` basic example, move the mouse to the preview screen and click the left and right buttons multiple times, try holding both buttons pressed and move the mouse, and you'll see "the underneath" of your model. It's a bug and looks like this in the preview

[image]

and rendered

[image]

<details>
<summary>OpenSCAD library info</summary>

```
OpenSCAD Version: 2022.09.16.ai12152 (git 2fd5842c9)
System information: Linux 5.15.0-47-generic #51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 x86_64 Ubuntu 22.04.1 LTS 4 CPUs 15.57 GB RAM
User Agent: OpenSCAD/2022.09.16.ai12152 (git 2fd5842c9) (Linux x86_64; Ubuntu 22.04.1 LTS)
Compiler: GCC "7.5.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_65_1
Eigen version: 3.3.4
CGAL version, kernels: 5.4, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.2
QScintilla version: 2.10.2
InputDrivers: 
GLib version: 2.56.4
lodepng version: 20210627
libzip version: 1.1.2
fontconfig version: 2.12.6 (runtime: 2.13.1)
freetype version: 2.8.1 (runtime: 2.11.1)
harfbuzz version: 1.7.2 (runtime: 2.7.4)
cairo version: 1.15.10
lib3mf version: 1.8.1
Features: fast-csg*, fast-csg-trust-corefinement*, fast-csg-debug, fast-csg-exact, fast-csg-exact-callbacks, fast-csg-remesh, fast-csg-remesh-predictibly, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, sort-stl
Application Path: /tmp/.mount_openscspW64U/usr/bin
Documents Path: /home/dizcza/.local/share
User Documents Path: /home/dizcza/Documents
Resource Path: /tmp/.mount_openscspW64U/usr/share/openscad
User Library Path: /home/dizcza/.local/share/OpenSCAD/libraries
User Config Path: /home/dizcza/.config/OpenSCAD
Backup Path: /home/dizcza/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/home/dizcza/.local/share/OpenSCAD/libraries
/tmp/.mount_openscspW64U/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/dizcza/.local/share/fonts
/home/dizcza/.fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/home/dizcza/Library/Fonts
/usr/share/fonts/X11
/usr/share/fonts/cMap
/usr/share/fonts/cmap
/usr/share/fonts/opentype
/usr/share/fonts/truetype
/usr/share/fonts/type1
/usr/share/fonts/X11/Type1
/usr/share/fonts/X11/encodings
/usr/share/fonts/X11/misc
/usr/share/fonts/X11/util
/usr/share/fonts/cmap/adobe-cns1
/usr/share/fonts/cmap/adobe-gb1
/usr/share/fonts/cmap/adobe-japan1
/usr/share/fonts/cmap/adobe-japan2
[... 432 lines cut ...]
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
```

</details>

jordanbrown0, 2022-09-18:
Possible duplicate of #4332.  If so, fixed by #4333.

dizcza, 2022-09-18:
> Possible duplicate of #4332. If so, fixed by #4333.

Indeed. I had a look at similar issues but found none at the time. You can close it as a duplicate.

thehans, 2022-09-18:
Duplicate of #4332

closed 2022-09-18 by thehans