openscad/openscad #4359: Preview is blank (black) with no errors

iansan5653, 2022-09-20:
Hello! I'm not sure if this is a bug or if it's me doing something obviously wrong (hopefully it's my fault and it's easy to fix 😄).

I'm working on trying to get OpenSCAD running in a [Codespace](https://github.com/features/codespaces). If you're unfamiliar, Codespaces are essentially just lightweight virtual machines with built-in VSCode integration. I know this isn't an environment that you officially support, but I'm not completely sure if this problem is specific to Codespaces or not. I'm hoping it's just some missing graphics library that I need to install.

## The Problem

OpenSCAD is running in the virtual environment and almost everything looks great, but the rendered preview pane is always a solid black rectangle:

[image]

Nothing appears in the error log and the console looks completely normal:

```
OpenSCAD 2021.01
https://www.openscad.org/

Copyright (C) 2009-2021 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.

Loaded design '/workspaces/openscad-template/box.scad'.
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 728
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.018
```

Here's my library & build info, collapsed for brevity:
<details>
<summary>Library and Build Information</summary>

```
OpenSCAD Version: 2021.01
System information: Linux 5.4.0-1090-azure #95~18.04.1-Ubuntu SMP Sun Aug 14 20:09:27 UTC 2022 x86_64 Debian GNU/Linux 11 (bullseye) 4 CPUs 7.77 GB RAM
User Agent: OpenSCAD/2021.01 (Linux x86_64; Debian GNU/Linux 11 (bullseye))
Compiler: GCC "10.2.1 20210110" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.3.9
CGAL version, kernels: 5.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.2
QScintilla version: 2.11.6
InputDrivers: 
GLib version: 2.66.4
lodepng version: 20180910
libzip version: 1.7.3
fontconfig version: 2.13.1
freetype version: 2.10.4
harfbuzz version: 2.7.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Application Path: /usr/bin
Documents Path: /home/vscode/.local/share
User Documents Path: /home/vscode
Resource Path: /usr/share/openscad
User Library Path: 
User Config Path: /home/vscode/.config/OpenSCAD
Backup Path: 
OPENSCADPATH: <not set>
OpenSCAD library path:

/usr/share/openscad/libraries

[... 296 lines cut ...]
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

</details>

## Reproducing

I'm running OpenSCAD 2021.01 on a Debian distribution using the following simple Dockerfile:

```dockerfile
FROM mcr.microsoft.com/vscode/devcontainers/base:0-buster

RUN apt-get update && export DEBIAN_FRONTEND=noninteractive \
    && apt-get -y full-upgrade \
    && apt-get -y install --no-install-recommends openscad
```

I'm viewing the OpenSCAD window through the [desktop-lite](https://github.com/devcontainers/features/tree/main/src/desktop-lite) devcontainer feature, which adds a desktop environment to codespaces, powered by the [Fluxbox](http://fluxbox.org/) window manager.

I've published this configuration as [a repository](https://github.com/iansan5653/openscad-template), so this can be reproduced by following the "Getting started" section in that repository's README.

Please let me know if there is any more info I can provide!

t-paul, 2022-09-20:
Getting only 404 on that repository, is it set to private?

iansan5653, 2022-09-20:
Fixed! Sorry about that

iansan5653, 2022-09-20:
I did see [this potentially related issue](https://github.com/openscad/openscad/issues/3494) which describes a similar problem, but I don't think it's the same issue - I don't have OpenGL ES installed and I don't see an error in the error log.

thehans, 2022-09-20:
The  bits per channel in your library info are concerning to me, `RGBA(5650), depth(0)`.  I think all our code expects 8 bits per color channel, and at least some depth buffer.  Not sure if that would cause this type of issue or not.  

I've never messed with codespaces, but one thing that's always good to try with OpenGL issues like this, is if you can get output from *any* GL program, such as `glxgears`.  If you can't, then it's likely a system config issue, and not OpenSCAD's fault ;)

iansan5653, 2022-09-20:
I ran `glxgears` (after installing `mesa-utils`) and it seems to run fine (at 1500 fps 👀), so it doesn't _look_ like a system issue:

[image]

t-paul, 2022-09-20:
Yeah, I was wondering about the 565 color support. We do need depth and stencil buffers, but that reporting seems unreliable. On my local machine (Dell XPS 13 running Debian/Testing) I'm getting:
```
GLEW version: 2.2.0
OpenGL Version: 4.6 (Compatibility Profile) Mesa 22.2.0-rc3
GL Renderer: Mesa Intel(R) UHD Graphics 620 (KBL GT2)
GL Vendor: Intel RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(0), stencil(0)
```

But with completely black screen, it's more likely a context issue. I suspect it's selecting a 3.1 forward compatible / core profile but we need the compatiblity for the ancient fixed pipeline stuff.

Maybe try running with:
```
MESA_DEBUG=context MESA_GL_VERSION_OVERRIDE=3.1COMPAT openscad
```
That should select the compat profile and output issues setting up the context.

iansan5653, 2022-09-20:
> Maybe try running with:
> 
> ```
> MESA_DEBUG=context MESA_GL_VERSION_OVERRIDE=3.1COMPAT openscad
> ```

No dice there 😞. I don't see any difference at all in UI, console output, or library & build info.

t-paul, 2022-09-20:
Hmm, and with even more variables added (second one probably not needed):
```
GALLIUM_DRIVER=softpipe LIBGL_ALWAYS_SOFTWARE=true
```
That might make everything extremely slow with the softpipe rendering but just to see if it makes a difference. Not having high hopes though as llvmpipe is usually quite good nowadays and I have no idea if they maybe even share the context setup code.

iansan5653, 2022-09-20:
Still no change with all four arguments. The rendering time is much slower (around 1.25s instead of around 0.25 seconds) but nothing appears still.

Small note - I realized in my original issue description and in the repo I wrote Debian `buster` instead of `bullseye`, but my screenshots & logs have been from `bullseye` (otherwise they'd be from the old 2019 version of OpenSCAD). I've fixed that now.

iansan5653, 2022-09-22:
Any other ideas I could try before I give up on this?

Is there a document with all of the environment variable options anywhere? I could just work through them and try anything that might affect it.

Or is there any way to get more detailed logging output? I feel like if I could just find one clue as to why this is happening, I might be able to figure it out. But I have absolutely no idea at this point.

t-paul, 2022-09-22:
That's all Mesa settings, full list can be found at https://docs.mesa3d.org/envvars.html

I'm not sure if there's any additional Qt setup possible, but that likely needs code changes. OpenSCAD itself has no options for tuning the OpenGL context creation.

As that's mostly driver interaction, I don't know if there's additional log possible other than what the Mesa environment variables enable. A more complicated by much more detailed trace would be possible using tools like `renderdoc` which follows all OpenGL calls.

I don't see that I have any access to codespaces. Maybe there's some way to do this locally but that might be more time consuming than what I can manage currently. Or is there dedicated documentation on mirroring the whole setup locally? I did run OpenSCAD via novnc docker container a while ago and that worked without problems.

iansan5653, 2022-09-22:
Thanks! That link at least gives me something else to play with.

> I don't see that I have any access to codespaces. Maybe there's some way to do this locally but that might be more time consuming than what I can manage currently. Or is there dedicated documentation on mirroring the whole setup locally?

I'm sorry, I forgot Codespaces is still in limited access beta. I'm a GitHub staff member so I may be able to get you access, but I'm not 100% sure as I'm not on that team. Working on it!

It's also possible to run locally, although I totally understand you may not have the time:

1. Install [Docker Desktop](https://www.docker.com/products/docker-desktop/) on your machine and start it so it's running in the background
2. Install [VSCode](https://code.visualstudio.com/) and the [Remote - Containers](https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.remote-containers) extension (`vscode-remote.remote-containers`)
4. Clone the repo locally (`https://github.com/iansan5653/openscad-template.git`) and open it in VSCode
5. When you open the repo, you may be prompted by the extension since there's a devcontainer configuration - if so, click "**Reopen in container**". If not prompted, run the VSCode command "**Remote-Containers: Reopen in container**"
6. You'll now be in the same environment that you would be if you were running a Codespace and can follow the instructions in the repo's readme to get started. Copying here for reference:
  1. View the virtual desktop in your browser at http://localhost:6080/
  2. Sign in with password `vscode`
  3. Back in VSCode, open `box.scad` and press the "Preview in OpenSCAD" button in the top-right (same icon as the preview button in OpenSCAD)
  4. The OpenSCAD window should appear on the virtual desktop in your browser. Editing box.scad in VSCode will automatically update the file in the OpenSCAD application

iansan5653, 2022-09-22:
✨ @t-paul You should have access now! You should be able to see a "Codespaces" tab in the **Code** dropdown on the repo:
[image]

iansan5653, 2022-09-23:
The only thing I've been able to find so far from playing with the environment variables is if I enable `LIBGL_DEBUG=verbose`, I see the following:

```
libGL: MESA-LOADER: dlopen(/usr/lib/aarch64-linux-gnu/dri/swrast_dri.so)
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/vscode/.drirc: No such file or directory.
```

But as far as I can tell, that doesn't look like a critical issue that would cause this problem.

iansan5653, 2022-09-23:
> I'm a bit surprised about the AARCH64. Are those 64-bit ARM machines running the code spaces?

This is using Docker Desktop on an M1 Max MacBook Pro. In Codespaces it's different:

```
libGL: MESA-LOADER: dlopen(/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/vscode/.drirc: No such file or directory.
```