openscad/openscad #4360: how to update macports version or get the dmg version working in MacOS via command line

WolfgangFahl, 2022-09-21:
https://ports.macports.org/port/openscad/ show Version 2015-03-03 and can't be installed due to 
```
parser.tab.c:145:10: fatal error: 'parser.tab.h' file not found
:info:build #include "parser.tab.h"
:info:build          ^~~~~~~~~~~~~~
```
in my mac OS Big Sur environment
I have installed the dmg version as a workaround but don't know how to use it from the command line to be used as part of a python library. 

How could i use OpenSCAD on my mac os with macports via command line?

WolfgangFahl, 2022-09-21:
see also https://trac.macports.org/ticket/65881

WolfgangFahl, 2022-09-21:
The workaround seems to be
```
/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD -v
OpenSCAD version 2021.01
```

t-paul, 2022-09-21:
Sorry the MacPorts build is not maintained here, so the ticket you created at macports.org seems to be the best way for reaching the right people.

Yes otherwise that script should work, I'd sugest adding quotes to `"$@"`. That should make it safe against parameters with spaces in them.

WolfgangFahl, 2022-09-22:
@t-paul  thx. I have also added a hint to https://wiki.bitplan.com/index.php/PythonScad#Installation. ViewScad now works properly with OpenSCAD which is great.

closed 2022-09-22 by WolfgangFahl