openscad/openscad #4361: Zoom with mouse wheel buggy

UserCain, 2022-09-23:
Version: 2021.01
Windwos 8.1

Zooming the viewport using the mouse wheel is not reliable.
Some times it work, but most of the time it zooms the editor window.
Also CTRL++ and CRTL +- affects the editor window only. 

Zoomig with Shift + Right Mouse Button **works**

The size of the view window dosen't matter.
Visibility of Axis doesen't matter.

jordanbrown0, 2022-09-23:
Zooms the editor window?  Mouse wheel alone should never zoom the editor window.
Mouse wheel alone should zoom the 3D view or scroll the editor or console, depending on where the mouse pointer is.

Ctrl+Wheel zooms whichever of the three the mouse cursor is in.

jordanbrown0, 2025-08-16:
Nor reproduced, no response.