openscad/openscad #4363: Cannot import SVG file containing a single path

jlpettersson, 2022-09-28:
I created a simple SVG file with a single `<path>` element containing a Bezier curve `C` and that is closed using the `Z`.

I was expecting that I could import this SVG file using `import("my-file.svg");`

But I get the error: 

    ERROR: Can't open file '/Users/jonas/Documents/airfoil/my-airfoil.svg', import() at line 1

This looks like a bug? At least the error-message should be more specific?

jordanbrown0, 2022-09-28:
Yes, it should be better.  Check for an error on stderr; it looks like there may be a better one there.

jordanbrown0, 2022-09-28:
If you attach a copy of the SVG I'll take a look at it.

jlpettersson, 2022-09-28:
This is the SVG file I tried with:

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   width="106mm"
   height="22.55mm"
   viewBox="0 0 106 22.55"
   version="1.1"
   id="svg399"
   inkscape:version="1.2.1 (9c6d41e, 2022-07-14)"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:svg="http://www.w3.org/2000/svg">
    <path
       style="fill:none;stroke:#000000;stroke-width:0.264583px;stroke-linecap:butt;stroke-linejoin:miter;stroke-opacity:1"
       d="M 2.9620763,10.988347 C 2.1976686,6.1152538 29.832237,1.986299 43.666735,3.0576269 48.699027,3.2815447 59.974283,3.9734222 82.460377,7.357415 86.664617,8.312924 118.5618,12.745737 95.264193,10.701695 90.97663,10.32552 88.066618,10.895031 80.644913,11.752754 63.273966,13.482875 37.279078,16.251809 14.428177,14.237075 2.6613521,13.882284 2.9620763,10.988347 2.9620763,10.988347 Z"
       id="path625"
      />
</svg>
```

t-paul, 2022-09-28:
Where is the file actually located, and do the filenames match (they don't in the import statement you gave).

jlpettersson, 2022-09-28:
Yes, my bad. It was a typo in the file-name. Thanks for your time.

closed 2022-09-28 by jlpettersson

t-paul, 2022-09-28:
I'm also seeing an error message due to the undefined `inkscape` prefix, but it seems to work anyway:
```
/tmp/x.svg:10: namespace error : Namespace prefix inkscape for version on svg is not defined
```

jordanbrown0, 2022-09-28:
Should still probably have a better error message.  It looks like libxml2 can provide more information - no-such-file versus access-denied versus et cetera.