openscad/openscad #4364: Feature Request: Keep file tabs when editor is hidden

geoffder, 2022-09-28:
When using an external editor or generating `.scad` files, it's nice to be able to hide the editor, but doing so currently also hides the file tab bar. Thus most of the time, I just minimize it as much as possible so that I can still see what it open and switch between them directly without having to TAB cycle. Would it be possible to decouple the tab bar from the editor for this purpose?

jordanbrown0, 2022-09-28:
When the editor is undocked, should the tab bar be with the editor window, or with the 3D view window?

jmckeown2, 2025-01-13:
> When the editor is undocked, should the tab bar be with the editor window, or with the 3D view window?

Why not have it independently dockable? 

But given that external editors are supported, and I’m not aware of a use case for having multiple editors open without a 3D view window, then keeping the tab bar with the 3D view is most logical.

closed 2025-03-09 by jordanbrown0