openscad/openscad #4365: Line wrap leading to significant slowdown (with long lines) even when editor is hidden

geoffder, 2022-09-28:
My use case is a little unusual, as I generate `.scad` files (with another language) made up in large part by polyhedron meshes. Whenever I would crank up the resolution of said generated meshes I would hit long file load times (time from opening/change until compile and preview). Until yesterday I dumbly assumed this must have been due to long parse (/ translation into internals) times of the several megabyte long `.scad`s, but while I was fiddling with other settings, it occurred to me to try turning line wrapping off in the editor. Turns out that was it all along! The editor was simply busy wrapping my absurdly long lines of points and faces whether it was hidden or not.

If the (potentially expensive) line wrapping could be avoided entirely while the editor is hidden it would certainly help with step around this problem (though, related to https://github.com/openscad/openscad/issues/4364 this would mean losing file tabs), but I wonder if there could also be some kind of a stopgap where trouble cases are detected and wrapping is skipped so the GUI doesn't get deadlocked (have experienced in excess of a minute) by default when such a file is loaded. I appreciate that I'm in a bit of a minority using OpenSCAD in a way that would ever run into this issue though.

thehans, 2022-09-28:
If you are generating `polyhedron` input from outside of OpenSCAD, then why not just write straight to STL and `import` instead?

geoffder, 2022-09-28:
> If you are generating `polyhedron` input from outside of OpenSCAD, then why not just write straight to STL and `import` instead?

I prefer to have the `.scad` files free of dependencies when possible, and aside from preventing me from coming across this issue, I'm not sure what benefits I would net by saving unrendered/verified intermediary STLs.

t-paul, 2022-09-28:
I'm not sure about adding complicated heuristics around the editor component (which we have not much control over as it's an external library) for a very special use case. What happens if you also generate some line breaks instead of a single line?

jordanbrown0, 2022-09-28:
Could also generate the .SCAD data into myobject.scad, but have the file that you load into the editor just be `include <myobject.scad>` so that you never load the huge file into the editor.

geoffder, 2022-09-28:
> I'm not sure about adding complicated heuristics around the editor component (which we have not much control over as it's an external library) for a very special use case. 

Understandable, I felt I should open the issue anyway rather than keep it to myself anyway, especially considering how the line wrap culprit didn't jump out at me until it did. It isn't too hard to warn potential users of my code to remember to turn off line wrap, but it does feel bad to have it happen without realizing what is going on.

> What happens if you also generate some line breaks instead of a single line?

I have tried introducing a line break after each vector/face element and it does indeed improve the situation, but there was still some lag compared to when line wrapping is turned off. I think owing to the fact my editor is usually fully "minimized", so even the short lines (of which there are many) can cause wrapping to engage. Also, not that it is the end of the world, but it does increase the `.scad` file size.

geoffder, 2022-09-28:
> Could also generate the .SCAD data into myobject.scad, but have the file that you load into the editor just be `include <myobject.scad>` so that you never load the huge file into the editor.

Sure, but given that there does exist a fix in the settings, I don't think I'm in a position that warrants the extra indirection.