openscad/openscad #4366: OPENSCADPATH not adhered to. Windows 10.

robiwano, 2022-09-30:
I've set the OPENSCADPATH variable to point to a specific location, but OpenSCAD does not recognize it and always opens the default location "C:\Users\<user>\Documents\OpenSCAD\libraries"

MichaelAtOz, 2022-09-30:
Please log the top section of Help/Library-Info (showing paths).

What do you mean by `always opens the default location`? File/Open?? <- if so that is not what it is for, it is for libraries etc in such things as `include <>`.

robiwano, 2022-09-30:
Yes, File/Show Library Folder doesn't open OPENSCADPATH. But I see that it IS adhered to, as the includes work as expected. Closing issue.

closed 2022-09-30 by robiwano

MichaelAtOz, 2022-09-30:
Hmm, perhaps, in an optimal future, `File/Show Library Folder` should offer-up the various library types for selection.

robiwano, 2022-09-30:
Or, maybe move the library folder selection into Preferences?

MichaelAtOz, 2022-09-30:
They all serve various purposes. 
Have you read the [wiki](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Libraries)?

nophead, 2022-09-30:
Preferences don't affect the command line but environment variables do, so that is the only mechanism to affect the GUI and the command line.

jordanbrown0, 2022-09-30:
Doesn't make it any more natural for the GUI, and since the GUI is probably 90%+ of users...

robiwano, 2022-10-01:
An idea is to add an editor for the OPENSCADPATH variable inside the Preferences pane. Best of both worlds ;)

jordanbrown0, 2022-10-01:
Can't really be done for Linux and similar environments.  There's no one place where environment variables are set - you set them somewhere in one of your startup scripts.

robiwano, 2022-10-01:
> Can't really be done for Linux and similar environments. There's no one place where environment variables are set - you set them somewhere in one of your startup scripts.

That is true. But, it is not uncommon for multiplatform tools to enable certain features depending on the platform. 

But, having a read-only field in Preferences just showing the value of OPENSCADPATH would go a long way.

MichaelAtOz, 2022-10-11:
Did you look at Help/Library-info?

 

...

Application Path: C:/Program Files/openscad-2021.01
Documents Path: C:\Users\...
User Documents Path: C:\Users\...
Resource Path: C:/Program Files/openscad-2021.01
User Library Path: C:/Users/..../libraries
User Config Path: C:\Users\...AppData\Local/OpenSCAD
Backup Path: C:/Users/...OpenSCAD/backups
==>> OPENSCADPATH: C:\Users\...\OpenSCAD Libraries;C:\Users;C:\test;C:\test2;C:\test-uc-?
OpenSCAD library path:
C:/Users/.../OpenSCAD Libraries
C:/Users
C:/test
C:/test2
C:/test-uc-?
C:/Users/.../OpenSCAD/libraries
C:/Program Files/openscad-2021.01\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/Windows/fonts
C:/Users/.../.local/share/fonts
C:/Users/.../.fonts

...

 

  _____  

 

 

This email has been checked for viruses by AVG antivirus software.
www.avg.com

MichaelAtOz, 2022-10-11:
Did anyone else get some email from history?

Below is in my sent mail dated 1 Oct 2022.

Plus three others.

 

  _____