openscad/openscad #4374: [FeatureRequest] Support additional and configurable 3D printing methods

lucafaggianelli, 2022-10-06:
It would be nice to automatically export the model to STL and send it to Cura (well this is my usecase :) or other slicers or printing methods.

How I envision this is that the user sets its own preferred printing method in the preferences and then when you hit the print button in the UI it just prints.

The print methods could be already predefined among the most used software, but it could be also possible to let the user launch a custom executable or program, i.e.: launch Cura to print.

Happy to take care of the development of this feature if anyone is interested.

tcurdt, 2022-10-11:
On macOS it could be the equivalent of exporting to a tmp destination and then just calling `open` on it.
Not entirely sure how this would need to implemented on Linux or Windows.

jordanbrown0, 2022-10-12:
If one was willing to have it be "send to default slicer", it would be very similar on Windows.

jordanbrown0, 2022-10-12:
Where the "default slicer" is whatever program opens STL files.

nophead, 2022-10-12:
STL's open NetFabb on my system. I have multiple slicers.

jordanbrown0, 2025-03-08:
Edit/Preferences/3D Print now allows you to specify a local application.

[image]

closed 2025-03-08 by jordanbrown0