openscad/openscad #4375: [Bug] Selected tab is completely white

lucafaggianelli, 2022-10-06:
I have multiple tabs opened and the selected one is white with no close button available (I don't if there should be one actually):

[image]

I'm running OpenSCAD v2021.01 on macOS 12.2.1

MichaelAtOz, 2022-10-06:
Debian & Windows have a close button (on the right end of the tab), so imagine Mac should too.

lucafaggianelli, 2022-10-06:
Mmmm interesting so the lack of the close button is a bug too :)

t-paul, 2022-10-06:
Please check with the development snapshot, if that shows the same issue.

lucafaggianelli, 2022-10-20:
Ah great, indeed it works fine in the dev snapshot `2022.10.17`:
[image]

t-paul, 2022-10-20:
Thanks, closing as fixed then.

closed 2022-10-20 by t-paul