openscad/openscad #4383: [bug] object 3d - rendered instead of 2d

nuliknol, 2022-10-16:
this is for version version 2021.01 , default install for Ubuntu 22.04

when I hit "Render" button, I get the correct output:
[image]

but if I hit the "Save" button, I get it 3d-rendered
[image]

I shouldn't get a 3d shape because I didn't extrude it yet.

t-paul, 2022-10-16:
That's a limitation of the preview engine, it might see some improvements but the behavior can't be easily removed. See #1566.

closed 2022-10-16 by t-paul

MichaelAtOz, 2022-10-17:
You have Design/Automatic-reload-and-preview set.

 

When you save, it auto reloaded and previewed (see the console).

That is a bug, that it reloads on save, and IIRC has been fixed in the dev version.

 

The apparent extrude is actually the preview, the library OpenCSG can't do 2D, so it does 1 unit centred Z.

There have been proposals to make it thinner, but it will always be apparently 3D.

 

  _____  

 

 <https://user-images.githubusercontent.com/1125368/196048933-1d33b55e-aead-4034-b2e1-cbdec9affb41.png> render-only

 <https://user-images.githubusercontent.com/1125368/196048940-a9be8925-a709-4e16-976b-ce3d71c9654f.png> savebuttonhit

 

This email has been checked for viruses by AVG antivirus software.
www.avg.com

nophead, 2022-10-17:
I work around it by using linear_extrude(eps) , where eps = 1/128.