openscad/openscad #4386: [FeatureRequest]Code editor needs debuging tools

williamlugg, 2022-10-18:
The code editor needs debugging tools added such as the ability to step through code and set and remove breakpoints.  These tools would operate in a manor similar to the way they operate in other IDEs.

Scopeuk, 2022-11-14:
To start a discussion on this, debugging tools for openscad would probably behave "strangely" to someone familiar with procedural programming, the geometry is evaluate outside in to build the components.
I could potentially see merit in being able to step through the layers as transforms are added. it is currently possible to approximate this using the modifiers
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters
% make this part transparent in final render
! only this subtree
* turn off this subtree

The only thing approximating a debugger I could see in a standard software sense would be functionally equivalent to walking a ! back through the tree.

jordanbrown0, 2022-11-14:
You can certainly walk through the execution phase.  (But people would be confused by what order OpenSCAD does stuff in.)

Walking through the geometry processing is less clear.  I don't think it's as simple as placing an !.  For one thing, I don't know what ! does when you apply it to something that is in a loop.

Also remember that geometry is a tree, not a linear sequence.  What would it mean to walk through this?
```
$fs=0.3;

difference() {
    union() {
        cube(10, center=true);
        sphere(6);
    }
    union() {
        cylinder(h=15, d=3, center=true);
        rotate([90,0,0]) cylinder(h=15, d=3, center=true);
        rotate([0,90,0]) cylinder(h=15, d=3, center=true);
    }
}
```
?

closed 2026-05-24 by github-actions[bot]