openscad/openscad #4388: Feature request: Support JPEG XL (.jxl) for exports

C0rn3j, 2022-10-19:
JPEG XL is a new-generation image format that I would love to see OpenSCAD support for exports.
https://jpegxl.info/

There is a massive difference in size even when losslessly converting a PNG generated by OpenSCAD to JXL:
[image]

t-paul, 2022-10-20:
At this point this might only be for Linux, all other don't seem to have the library readily available in their repositories (macOS Homebrew, MXE for Windows). For our macOS builds, this would need someone to help with setting up a universal build.

t-paul, 2022-11-01:
With Google (as owner of the CLA) [dropping support](https://www.theregister.com/2022/10/31/jpeg_xl_axed_chrome/) due to *"...not enough interest from the entire ecosystem to continue experimenting with JPEG XL"*, I'm closing this for now. In case of growing support, and hopefully better availability in distributions, we can always revisit the topic.

C0rn3j, 2022-11-01:
I do not believe Alphabet axing support in Chromium with very dubious 'there is no interest' claim should have any bearing on other projects.

https://bugs.chromium.org/p/chromium/issues/detail?id=1178058#c84

There is 70 people in the last 24 hours alone complaining about this decision on that bug thread, clearly there is a lot of interest and I believe I have demonstrated a valid use case.

t-paul, 2023-01-31:
Mozilla will not support JPEG XL at this time either https://github.com/mozilla/standards-positions/issues/522#issuecomment-1409539985

C0rn3j, 2023-06-08:
Apple will support JPEG XL. [1]

JPEG XL support keeps being added and improved in other software too. [2]

I ask that you reconsider and reopen the request.

[1] https://www.theregister.com/2023/06/07/apple_safari_jpeg_xl/

[2] https://www.phoronix.com/news/FFmpeg-Animated-JPEG-XL