openscad/openscad #4391: Reseed random number generator on each run

jordanbrown0, 2022-10-20:
The random number generator is currently seeded (with the current time plus the process ID) at application startup, and after that only when the program calls rands(x,x,x,seed).

This can yield odd behavior:

```
echo(rands(0,1,1));
echo(rands(0,1,1,1));
```
will, after the first run, always yield the same two numbers:  the second and first numbers in the sequence seeded by 1.

It would be more obvious if the first number was different each time.

The fix seems to be simple:  reseed at the start of each run.  To add a little spice, mix in a counter so that multiple runs in the same second don't get the same results.

MichaelAtOz, 2022-10-22:
Note, the 2021.01 release notes lists:

> Changed
>    Allow random seeds to stick between rands() calls

rcolyer, 2022-10-22:
> Note, the 2021.01 release notes lists:
> 
> > Changed
> > Allow random seeds to stick between rands() calls

Yeah, that was me.  I will certify that Jordan's proposal is compatible with the goals of that change, and an improvement to it.  The "seeds to stick" change had the motivation of making procedurally generated animations work elegantly by permitting a single seed at the top of the program to stick throughout all the subsequent calls to rand, but preserving a consistency from frame to frame yielding a consistent procedurally generated image based on that seed choice.  It was implemented by a global rng receiving a seed which sticks, and which complies with what you would need in most programs.  Persistence between runs was an unintended side-effect of the implementation that I recognized while implementing it, but that wasn't considered important at the time because there wasn't a clear use case available for why it should matter given how I saw rands being used at that time under the prior behavior.

Reseeding (if done correctly with a variable enough seed) at the start of each run provides the nuance to provide initial unseeded randoms that remain "random", and keeps the consistency after a seed within each run so animations continue to work with the persistence property correctly.

I will add a note for future people referring back, that a better eventual solution would look more like rng object construction with OS sourced random seed or specified seed, so that we can mix and match within a run.  But for now we still have this built-in rands function call that is proper to fix up the behavior of the way Jordan proposed.

jordanbrown0, 2022-10-22:
> Reseeding (if done correctly with a variable enough seed) at the start of each run provides the nuance to provide initial unseeded randoms that remain "random", and keeps the consistency after a seed within each run so animations continue to work with the persistence property correctly.

To be clear:  I believe that here you are referring to reseeding *by OpenSCAD itself*.  The OpenSCAD program has no source of entropy other than rands() and so no way to generate useful seeds.