openscad/openscad #4393: bug in 2D system

jon-bondy, 2022-10-23:
Here's a minimal program that demonstrates the problem:

    rotate_extrude() {
        square([3, 1]);
    	translate([2, 1])
    		square([1, 1]);
    	square([1, 2]);
    }
    translate([-100,0,0]) cube();

This appears to be a bug in the 2D subsystem that is then leading to CGAL problems.

Looking at just the 2D object, with F6, is illuminating:

    square([3, 1]);
    translate([2, 1])
        square([1, 1]);
    square([1, 2]);

Notice that the top-right square, the second one created, is not connected to the others.  Strangely, removing the left-hand one causes the right-hand one to be connected.

A workaround is to make the right-hand square start at Y=0 and be 8 tall instead of 6.

UBaer21, 2022-10-23:
can't see an issue with version 22  however the 1×1 square has no overlap as it should.

UBaer21, 2022-10-23:
removing the overlap also connects the other non overlapping square
```
union(){
  square([3, 1]);
	translate([2, 1])square([1, 1]);
	translate([0, 1])square([1, 1]);
}
```
[image]

jon-bondy, 2022-10-24:
I don't know what you mean by "version 22" - all of my OpenSCAD releases have dates as version numbers.

Also not sure what "has no overlap as it should" means: does this mean that there is a bug, or no bug?  Also not clear what you mean by "removing the overlap also connects the other non overlapping square"

UBaer21, 2022-10-24:
@jon-bondy sorry for the confusion, version 22 is meant to be year 2022 like any of the [nightly builds ](http://openscad.org/downloads.html#snapshots)
With overlap is meant that a[ general rule](https://en.wikibooks.org/wiki/OpenSCAD_Tutorial/Chapter_1#Adding_more_objects_and_translating_objects) in openSCAD is to have overlapping areas/volumes in unions.

The square has one segment on the same location, so it could happen due to the limitation of numbers in computers and possible grid that one square is moved very slightly away which causing a gap. 

here the code with 0.001 overlap
```
union(){
  square([3, 1]);
	translate([2, 1 - 0.001])square([1, 1 + 0.001]);
	translate([0, 1 - 0.001])square([1, 1 + 0.001]);
}
```

jon-bondy, 2022-10-24:
In general, when using implicit union on two 3D cubes, there is no need to do this kind of overlap.  That is, in about 5 years of using OpenSCAD extensively, I have never encountered this kind of problem in the 3D realm.  This is why I would consider it a bug in the 2D realm.  Do you agree?

nophead, 2022-10-24:
Yes in 2D integer coordinates should be exact because it uses fixed point 64 bit representation.  And even in CGAL I don't get problems stacking cubes if the coordinates are integers. It is only when floating point fractions are involved, like if you rotate a cube by 45 degrees. Then the coordinates are irrational so impossible to represent exactly even in rationals.

jon-bondy, 2022-10-24:
Is this difference in behavior between the 2D and 3D systems documented?  It is SURE to catch people up repeatedly.  But I guess it is all baked into the design of the system, and cannot be easily changed.  Strange that I have used OpenSCAD for so many years and never run into it before.  And I WAS using integers!

nophead, 2022-10-24:
I think it must be regression in the 2D system as I only noticed it recently. I used to find it rock solid.

UBaer21, 2022-10-24:
maybe it is the  const double GRID_FINE = 0.00000095367431640625; 
normally you only see problems on bigger objects with values over 1mill. 
e.g:
```
x=1000002;
translate([1,1,1]*x/2)cube(x,true);
echo(x,x/2);
```
[image]

t-paul, 2022-10-24:
Nope, that's a display resolution glitch. Slowly zooming in/out at this scale makes it jump around the origin on my screen.

Try exporting as ASCII-STL:
```
  facet normal -1 -0 0
    outer loop
      vertex 0 1e+06 1e+06
      vertex 0 0 0
      vertex 0 0 1e+06
    endloop
  endfacet
```

t-paul, 2022-10-24:
Hmm, maybe should try something with double precision. Also `cube()` is not a good example for 2D :grinning:.

kintel, 2024-04-01:
Status as of 2024-03-24: Works with Manifold, still broken with CGAL

kintel, 2024-12-20:
Fixed by #5501

closed 2024-12-20 by kintel