openscad/openscad #4394: Consider upgrading to Clipper2

thehans, 2022-10-23:
The author of the Clipper library (sometimes known as "polyclipping"), which we use for 2D booleans and offsets, has released a major update which I think would be worth looking into.
https://github.com/AngusJohnson/Clipper2

Currently we have source files of clipper included into our own project under `src/ext/polyclipping`.   With Clipper2 being in a github repo, I think it would be simpler and cleaner to build as a git submodule instead.

I'm not sure yet how much of the interface has changed, so no idea how large or small of an effort this would be.

thehans, 2022-10-23:
There is also a chance that the new version happens to fix some existing issues such as #4393

nophead, 2022-10-23:
I might be wrong but I think that might be a regression. I ran into  such problems in the last year maybe and probably in the last release but the 2D system seemed to be solid going way back before then. It took me by surprise when I was just subtracting squares with coincident edges. I am sure I didn't need an overlap in the past when the coordinates are exactly coincident.

PR #5501 (2024-12-20): Update to Clipper2
Updates OpenSCAD's usage of ClipperV1 to ClipperV2:
* Uses a submodule instead of forking the sources
* Allows using a system-provided Clipper2
* Allows using the same Clipper2 library as Manifold
* Changed precision to 10^8 to align with Manifold's 2D subsystem

Fixes #4394 #4974 #4393
closed 2024-12-20 by kintel