openscad/openscad #4398: Should .csg format store camera data?

thehans, 2022-10-26:
There is an issue with a test  named `csgpngtest_logo_and_text`.  This test is not part of the default test suite, so is not tested by CI servers, but can be tested manually with `ctest -C All`.  

The `csgpngtest` test type basically exports to .csg, then starts another instance of OpenSCAD importing the csg, and exports an image from that.  Finally this is compared with the expected image for `cgalpngtest`, which comes from a normal CGAL render to PNG.

The problem this causes for `logo_and_text.scad` is that it has `$vp*` variables set in the `.scad` file to define the camera position, that means if you generate expected image from `cgalpngtest_logo_and_test`, then imported .csg version fails.  

Alternatively if one were to update the expected results based on `csgpngtest_logo_and_text`, then the `cgalpngtest` will fail.

If its not something anyone cares very strongly about, then the quick fix for this would be to add `csgpngtest_logo_and_text` to the list of disabled tests in `tests/CMakeLists.txt`
https://github.com/openscad/openscad/blob/2d27703db5ac6f31ed37d2274d176668e940a502/tests/CMakeLists.txt#L741-L807

jordanbrown0, 2022-10-26:
I would say that we include color not because we want to support preview, but rather because color is part of the definition of the model.  The fact that it's only supported in preview is not a "frill" of preview, but rather a missing feature of render.

Note that we do not record places where you set $fn and its friends.  Rather, we record places where they are consumed.  `$fn =10; cube();` will not record $fn in the CSG tree.

jordanbrown0, 2022-10-26:
And circle() will record $fn and its friends whether or not you set them.