openscad/openscad #4404: cylinder() is actually a cone

AcinonXDev, 2022-11-02:
The function ```cylinder(h,r1,true);``` produces a cone, not a cylinder. Please close this issue if the function is actually meant to produce a cone instead and different arguments are used for a cylinder. But even in that case, it is still unintuitive to have the `cylinder()` function produce both a cylinder and a cone depending on the arguments.

nophead, 2022-11-02:
I think you want `cylinder(h,r1, center = true)`

I think the true is being taken as the second radius which would be almost a cone but with a flat point.

nophead, 2022-11-02:
This works: `cylinder(10, r = 3, center =true);`

If you use r2 I think you also need to specify r2 or ir will default to 1.

closed 2022-11-02 by AcinonXDev

jordanbrown0, 2022-11-02:
The [documentation](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#cylinder) hints at this potential confusion, but is itself confusing.  It says "Parameter names are optional if given in the order shown here", but does not clearly show an order.  There's a list of parameters further down, but that comment is *not* correct for that list.  You cannot say `cylinder(h, r)` and get what you expect; the second arg is taken as r1.  (And r2 defaults to 1.)

The cheat sheet is similarly confusing; it implies that you can just say `cylinder(h,r,center)` or `cylinder(h,d,center)`.  Obviously those can't both work, even if the conflict with r1/r2 is addressed.

I almost always specify arguments by name, not position, so would say `cylinder(h=5, r=20)`.

Cones are cylinders just like rectangular prisms are cubes and rectangles are squares :-)

jordanbrown0, 2022-11-02:
I think maybe the documentation intent is that "the order shown" is in the source sample three paragraphs down, but the intervening detailed text breaks that connection, at least for me.

jordanbrown0, 2022-11-02:
If you have questions, you might try the chat room.  There are people active there very nearly 24 hours per day.
https://openscad.org/community.html#irc
https://web.libera.chat/?channel=#openscad