openscad/openscad #4408: Windows CI Server has broken GL

thehans, 2022-11-04:
The Windows build on Github Actions started failing lately for all tests which involve rendering to PNG.

I have looked at a diff between two build logs to try to get an idea of what is going on:
OLD build with no failures [Windows build 1741](https://github.com/openscad/openscad/actions/runs/3357301535/jobs/5563020159)
NEW build with all images failing: [Windows build 1743](https://github.com/openscad/openscad/actions/runs/3357301535/jobs/5563020159)

One of the first things is that the runner has a new name:
```
OLD:  2022-10-30T22:23:34.6457372Z Job is about to start running on the hosted runner: Hosted Agent (hosted)
NEW:  2022-11-02T18:04:15.9865313Z Job is about to start running on the hosted runner: GitHub Actions 2 (hosted)
```

OLD build has a section which downloads mesa and llvm together:
```
2022-10-30T22:27:33.3216285Z Packages (2) mingw-w64-x86_64-llvm-14.0.6-6  mingw-w64-x86_64-mesa-22.1.7-1
2022-10-30T22:27:33.3216869Z 
2022-10-30T22:27:33.3217172Z Total Download Size:    77.23 MiB
2022-10-30T22:27:33.3217641Z Total Installed Size:  515.09 MiB
2022-10-30T22:27:33.3217962Z 
2022-10-30T22:27:33.3218339Z :: Proceed with installation? [Y/n] 
2022-10-30T22:27:33.3218820Z :: Retrieving packages...
2022-10-30T22:27:35.9416160Z  mingw-w64-x86_64-llvm-14.0.6-6-any downloading...
2022-10-30T22:27:35.9416702Z  mingw-w64-x86_64-mesa-22.1.7-1-any downloading...
```
NEW build has a slightly newer mesa version, and some additional SPIRV and OpenCL packages within that section:
```
2022-11-02T18:08:21.0047814Z Packages (7) mingw-w64-x86_64-clang-15.0.3-1  mingw-w64-x86_64-llvm-15.0.3-1  mingw-w64-x86_64-opencl-headers-2~2022.09.30-1  mingw-w64-x86_64-opencl-icd-2022.09.30-1  mingw-w64-x86_64-spirv-llvm-translator-15.0.0-1  mingw-w64-x86_64-spirv-tools-2022.3-1  mingw-w64-x86_64-mesa-22.2.2-1
2022-11-02T18:08:21.0048681Z 
2022-11-02T18:08:21.0048988Z Total Download Size:    138.04 MiB
2022-11-02T18:08:21.0049459Z Total Installed Size:  1028.43 MiB
2022-11-02T18:08:21.0049775Z 
2022-11-02T18:08:21.0050145Z :: Proceed with installation? [Y/n] 
2022-11-02T18:08:21.0050570Z :: Retrieving packages...
2022-11-02T18:08:31.0637019Z  mingw-w64-x86_64-llvm-15.0.3-1-any downloading...
2022-11-02T18:08:31.0638079Z  mingw-w64-x86_64-clang-15.0.3-1-any downloading...
2022-11-02T18:08:31.0638947Z  mingw-w64-x86_64-mesa-22.2.2-1-any downloading...
2022-11-02T18:08:31.0639636Z  mingw-w64-x86_64-spirv-tools-2022.3-1-any downloading...
2022-11-02T18:08:31.0640291Z  mingw-w64-x86_64-spirv-llvm-translator-15.0.0-1-any downloading...
2022-11-02T18:08:31.0640960Z  mingw-w64-x86_64-opencl-icd-2022.09.30-1-any downloading...
2022-11-02T18:08:31.0641569Z  mingw-w64-x86_64-opencl-headers-2~2022.09.30-1-any downloading...
```

The OLD build shows this MESA error at the start of tests (even though it passes all tests):
```
2022-10-30T22:51:49.0327871Z 'D:\a\_temp/msys64/mingw64/bin/opengl32.dll' -> './opengl32.dll'
2022-10-30T22:51:49.1286637Z running 'D:/a/openscad/openscad/build/tests/../openscad.exe --info' to generate sysinfo.txt
2022-10-30T22:51:50.1748946Z 
2022-10-30T22:51:50.1749484Z MESA: error: ZINK: vkCreateInstance failed
2022-10-30T22:51:50.1931611Z Post test:D:/a/_temp/msys64/mingw64/bin/python3.exe "D:/a/openscad/openscad/tests/test_pretty_print.py" --builddir=D:/a/openscad/openscad/build/tests 
2022-10-30T22:51:50.1953851Z Test project D:/a/openscad/openscad/build/tests
```

The NEW build does not print this error:
```
2022-11-02T18:33:15.2988534Z 'D:\a\_temp/msys64/mingw64/bin/opengl32.dll' -> './opengl32.dll'
2022-11-02T18:33:15.3771602Z running 'D:/a/openscad/openscad/build/tests/../openscad.exe --info' to generate sysinfo.txt
2022-11-02T18:33:17.0793809Z Post test:D:/a/_temp/msys64/mingw64/bin/python3.exe "D:/a/openscad/openscad/tests/test_pretty_print.py" --builddir=D:/a/openscad/openscad/build/tests 
2022-11-02T18:33:17.0811801Z Test project D:/a/openscad/openscad/build/tests
```

OLD build shows "llvmpipe" in report name
```
Testing/Temporary/win_8664_llvmpipe-_llvm-14.0.6,-256_twvc_report.html
```
NEW build shows "no_GL_renderer" in report name
```
Testing/Temporary/win32_no_GL_renderer_report.html
```

OLD test report system info:
```
Sysid: win_8664_llvmpipe-_llvm-14.0.6,-256_twvc

Result summary: 1520 / 1520 tests passed (100%)

System info

OpenSCAD Version: 2022.10.30
System information: Windows(TM) 10.0 SP 0.0 NTW 0 MSDN 724833 2 CPUs 7.00 GB RAM
User Agent: OpenSCAD/2022.10.30 (Windows(TM) 10.0 SP 0.0 NTW 0 MSDN 724833)
Compiler: GCC "12.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_80
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.5
QScintilla version: 2.13.3
InputDrivers: 
GLib version: 2.72.3
lodepng version: 20210627
libzip version: 1.9.2
fontconfig version: 2.14.0
freetype version: 2.12.1
harfbuzz version: 5.1.0
cairo version: 1.17.6
lib3mf version: (not enabled)
Features: fast-csg, fast-csg-trust-corefinement, fast-csg-debug, fast-csg-exact, fast-csg-exact-callbacks, fast-csg-remesh, fast-csg-remesh-predictibly, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, sort-stl
Application Path: D:/a/openscad/openscad/build
Documents Path: C:\Users\runneradmin\Documents
User Documents Path: C:\Users\runneradmin\Documents
Resource Path: D:/a/openscad/openscad
User Library Path: C:/Users/runneradmin/Documents/OpenSCAD/libraries
User Config Path: C:\Users\runneradmin\AppData\Local/OpenSCAD
Backup Path: C:/Users/runneradmin/Documents/OpenSCAD/backups
OPENSCADPATH: D:/a/openscad/openscad/tests/../libraries
OpenSCAD library path:
  D:/a/openscad/openscad/tests/../libraries
  C:/Users/runneradmin/Documents/OpenSCAD/libraries
  D:/a/openscad/openscad\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
  C:/Windows/fonts
  C:/Users/runneradmin/AppData/Local/Microsoft/Windows/Fonts
  D:/a/_temp/msys64/home/runneradmin/.local/share/fonts
  D:/a/openscad/openscad/build/tests/D/fonts
  D:/a/_temp/msys64/mingw64/share/;D/fonts
  D:/a/_temp/msys64/usr/local/share/;D/fonts
  D:/a/_temp/msys64/usr/share/fonts
  D:/a/_temp/msys64/home/runneradmin/.fonts
  D:/usr/X11/lib/X11/fonts
  D:/System/Library/Fonts
  D:/Library/Fonts
  D:/a/_temp/msys64/home/runneradmin/Library/Fonts

GLEW version: 2.2.0
OpenGL Version: 4.5 (Compatibility Profile) Mesa 22.1.7
GL Renderer: llvmpipe (LLVM 14.0.6, 256 bits)
GL Vendor: Mesa/X.org
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: WGL
PNG generator: lodepng
OS info: Microsoft(TM) Windows(TM) 10 0 20348 0xcfbfffec14 amd64
Machine: 8664
Git branch: master from origin https://github.com/openscad/openscad
Git upstream:
Image comparer: ImageMagick
```

NEW test report system info:
```
Sysid: win32_no_GL_renderer

Result summary: 345 / 1520 tests passed (23%)

System info

win32
system cannot create offscreen GL framebuffer object
system cannot create GL based images
Image comparer: ImageMagick
```

Scopeuk, 2022-11-04:
it looks like one of the issue links may be incorrect, should the new build be https://github.com/openscad/openscad/actions/runs/3391387123/jobs/5636510189 ?

thehans, 2022-11-04:
> it looks like one of the issue links may be incorrect, should the new build be https://github.com/openscad/openscad/actions/runs/3391387123/jobs/5636510189 ?

Thanks, I've edited my post to have the correct 2nd link. 
The one you linked is "new" and has the same failing issue, but I was trying to link the specific build which I had originally diff 'ed.

Scopeuk, 2022-11-04:
having a look at the "LastTest.log" generated in the artefacts it looks like the openscad exe is being killed by windows for an access violation during the opengl tests
return code 3221225477 / 0xC0000005  (see https://learn.microsoft.com/en-us/openspecs/windows_protocols/ms-erref/596a1078-e883-4972-9bbc-49e60bebca55?redirectedfrom=MSDN)

my best guess is that the mesa opengl dll is trying to call into a function in another dll making up part of the mesa engine, not finding it and tripping.
Depending on how our windows path variable is set we may not need to copy the current dll over to cause the override (if msys/mingw is before the windows system path) at which point omitting it would prevent any relative path issues manifesting.
I'll see if I can re-create this locally later

Scopeuk, 2022-11-05:
An additional work arround open to us maybe to bypass msys2 for Mesa.
The windows port maintainer has a pinned issue for a github action that does a system wide install https://github.com/pal1000/mesa-dist-win/issues/114 which refences to action https://github.com/ssciwr/setup-mesa-dist-win if nothing else that would let us pin a version of we see issues. The msys package is derived from the pal1000 repo

thehans, 2022-11-07:
That action seems to be working great for installing mesa 22.1.7 in PR #4409.  Thanks for pointing it out @Scopeuk.

A temporary test of using the same action to install 22.2.2 resulted in the tests freezing up on `cgalpngtest_child-modifier`
Before the freeze it was actually passing a majority of image based tests though, so a little different outcome from the msys2 mesa install.
There were also a few mysterious test failures before the freeze, which looked to be something font related, based on the test names.  But couldn't get any report since the build had to be cancelled.

Fix is merged with version hard-coded to 22.1.7 for now.
Will keep this issue open awaiting further updates on the upstream issue.

thehans, 2022-11-26:
Well, Windows broke again, but it is still on mesa 22.1.7, so its back to square one.  
No idea what is going on, but the CI environment did receive an update again.

I downloaded a log of before/after errors, stripped timestamps, and ran a diff to see what changed:
```
Image: windows-2022
Version: 20221027.1
Included Software: https://github.com/actions/runner-images/blob/win22/20221027.1/images/win/Windows2022-Readme.md
Image Release: https://github.com/actions/runner-images/releases/tag/win22%2F20221027.1
```

```
Image: windows-2022
Version: 20221120.1
Included Software: https://github.com/actions/runner-images/blob/win22/20221120.1/images/win/Windows2022-Readme.md
Image Release: https://github.com/actions/runner-images/releases/tag/win22%2F20221120.1
```
[Diff of `Windows2022-Readme.md` changes](https://github.com/actions/runner-images/commit/9b06b852013a0b207a0f73fc2b9aea75c29287f5)

Also a number of things were updated specifically in msys2.
`msys2-runtime 3.3.6-4` changed to `msys2-runtime 3.3.6-5`
And here is a diff of msys2 package versions (I cut out the packages which didn't change versions)
[image]

thehans, 2022-11-26:
I mistakenly made a comment (deleted now) that there was another mesa issue, but this time it was actually due to an error I introduced into `tests/CMakeLists.txt` by changing reference of `openscad.exe` to `openscad.com`