openscad/openscad #4412: Intersection is interacting with unexpected objects.

axle12693, 2022-11-08:
In the following toy example, I would expect the red cylinder with radius 1 not to be shown.

`$fn=100;

translate([0.5,0.8,0]) color("green") cylinder(0.2,1,1);
color("red") intersection() {
    translate([.9,0,0]) cylinder(0.2,0.2,0.2);
    cylinder(0.2, 1, 1);
}`

However, where it intersects with the green cylinder, it is shown. Is this a bug, or a fault in my understanding?

Using OpenSCAD 2021.01 on Windows 10 Home.

UBaer21, 2022-11-08:
I got
[image]
you may see z-fighting so while the red and green share the same plane you see the color bleeding through.
You can use render() (after the color) to calculate the points in preview.

Scopeuk, 2022-11-08:
trying this I am seeing z fighting at angles but it occurring where it should not, the colour modifier appears to be causing elements that were "removed" by the intersection to show as z fighting over the green shape
[image]

swapping the order to
`$fn=100;
color("red") intersection() {
translate([.9,0,0]) cylinder(0.2,0.2,0.2);
cylinder(0.2, 1, 1);
}
translate([0.5,0.8,0]) color("green") cylinder(0.2,1,1);
`

resolves the visual issues. this looks like a bug to me.

UBaer21, 2022-11-08:
@Scopeuk  Sorry should have clarified that i used an z-offset to get rid of the z-fighting - i thought the bug was about wrong geometry .. those z-fighting happens all the time with all objects involved when not using render.

Scopeuk, 2022-11-08:
the z fighting is expected, the issue is the size of the red component, it should be much smaller, we appeared to have colour coded it's constituent parts rather than the result of the intersection

UBaer21, 2022-11-08:
when coloring the intersection objects this can be seen 
[image]
```
$fn=36;
translate([0.5,0.8,0]) color("green") cylinder(0.2,1,1);
intersection() {
color("red")translate([.9,0,0]) cylinder(0.2,0.2,0.2);
color("blue")cylinder(0.2, 1, 1);
}
```

thehans, 2022-11-08:
Duplicate of #1000

closed 2022-11-08 by thehans