openscad/openscad #4413: Export image from precomputed geometry

minad, 2022-11-11:
For [scad-mode](https://github.com/openscad/emacs-scad-mode) I continuously export images for preview, when moving the camera. Is it possible to precompute the geometry once and cache it, such that camera movements don't require a full rerendering the next time openscad is called?

minad, 2022-11-11:
When I export first scad->csg and then in the second step csg->png I can get some speedup for expensive computations. However the second export still takes significant time. Is there a possibility to speed this up even more, similar to when I move the camera with the mouse in the OpenSCAD gui?

jordanbrown0, 2022-11-11:
Could you describe in more detail exactly what you're doing?  Are you invoking the openscad program multiple times, or are you making an animation, or...?

minad, 2022-11-11:
@jordanbrown0 It like an animation. I invoke openscad multiple times on the same scad file for each rendered frame when the user moves the camera in the Emacs GUI. The rendering of each frame shouldn't require recomputing the geometry every time. It seems the scad->csg computation is the best one can do?

Of course I don't expect best performance here. Going the route via intermediate images is necessarily inefficient. For best performance one should just use the native OpenSCAD GUI. It is nice nevertheless to display the rendered/previewed object as image directly in Emacs.

jordanbrown0, 2022-11-12:
There are three performance costs in that picture:
* Overhead to start OpenSCAD.
* Cost to execute the OpenSCAD-language program (yields CSG tree)
* Either:
  * Preview cost
  * Render cost

For most models, the cost to execute the OpenSCAD program is very small.  In theory you could do this once, and then move the camera around as you like.

Preview cost is generally low, but is camera-dependent.  You have to redo the preview work if you move the camera.

Render cost is generally high, usually the most expensive component by orders of magnitude.  The result is (roughly) an STL file, which you could then feed through preview (and the preview would be cheaper).

What you really need is an OpenSCAD that runs in kind of a daemon mode, where you can give it new files to execute or new camera positions, like you can interactively, without having to reload OpenSCAD, rerun the OpenSCAD-language program, et cetera.  But we don't have that.

MichaelAtOz, 2022-11-12:
Did anything ever come from the file based cache thing I vaguely recall from some time ago?

 

  _____  

 

 

This email has been checked for viruses by AVG antivirus software.
www.avg.com

MichaelAtOz, 2022-11-12:
Persistent Cache #2214 & #1645 have not progressed much.
This could be viewed as a duplicate but leaving this open as there may be other solutions.

@minad is this just for preview on screen or do you actually want an exported file? 
If just preview, a hack could be to have an OpenSCAD instance with auto-reload-preview, have that instance off-screen and scrape the preview window, that would be straight forward with AutoHotkey. You could also drive the camera via appropriate customizer use.

minad, 2022-11-12:
@jordanbrown0 Thank you for the precise explanation of the costs. Yes, a daemon mode would be exactly what I need. Is there any chance that something like this could be added? But I doubt that there are many use cases beyond my own. The offscreen solution described by @MichaelAtOz goes in that direction, but it would be highly platform dependent.

MichaelAtOz, 2022-11-12:
I was about to suggest adding a key shortcut to copy the preview, but `Edit/Copy_Viewport_Image` already exists.
It would clobber the clipboard, unless you can do some equivalent to a critical-section in the IDE, while saving/restoring the clipboard on retrieving the image. 
You'd want some things to help, like some of my suggestions in #2845, particularly some activity indicator (busy-signal/icon/status-bar_text etc) to let you know that longer running activity (F5/6 or export etc) has finished.
I had that basically working, but without a few helpers it will have limitations.

minad, 2022-11-12:
@MichaelAtOz You are on Windows right? I am on Linux/X11 but I would need a platform independent solution for the off-screen rendering and screen grabbing. I don't know how to do this. Also regarding the Unix perspective it doesn't feel right to do this. A daemon mode as mentioned by @jordanbrown0 would be a saner solution.

What do you use the screen grabbing for? Animations? Automation?

MichaelAtOz, 2022-11-12:
I have a strong Windows background & was fluent in AutoHotkeys (now a bit rusty, but it comes back easily), my legacy host (with email & day-to-day stuff) is Windows (a VM) but I've been using Debian (hosting the VM) for a while too, but still on the learning curve.

I wrote it for a challenge, basic Automation, and possibly to run some Windows based OpenSCAD GUI test scripts, as there are no GUI tests ATM. But it is stalled, it really needs a status mechanism, and ideally getting accessibility working for menus, so the OpenSCAD window doesn't need to be active/foreground.

If someone wants to suggest a linux equivalent to AutoHotkeys (the screen scrape, keyboard injection & maybe accessibility aspects), I'd give it a look.

thehans, 2022-11-13:
OpenSCAD is theoretically controllable much like a daemon, via the [D-Bus interface](https://github.com/openscad/openscad/blob/master/src/gui/input/DBusInputDriver.h).
There is already support for dbus as an input method for controlling the camera and performing "actions" (pretty much anything you can do from the GUI menu).  

See Edit -> Preferences -> Axes tab, to enable DBus checkbox.  
You can view the list of actions in the dropdowns of the Preferences -> Buttons tab.

I have never used the dbus interface myself though.

minad, 2022-11-13:
@thehans The dbus interface looks interesting! You are right that with this OpenSCAD will act like a daemon.