openscad/openscad #4417: Can't figure out what is a problem (can't compile, binary doesn't detect system color theme)

iLikeTrioxin, 2022-11-14:
Hi, i had a problem with theme, I heard that openscad should detect system theme and apply it but it does not. So i decided as with most of software on arch if binary doesn't work try compiling, but i can not get it to compile.

This is my trial to install openscad-git from AUR using yay:
https://pastebin.com/q1658jmH

This is output from manual compilation using "Building for Linux/BSD":
uni-get-dependencies.sh : https://pastebin.com/WdRWDBHQ
check-dependencies.sh    : https://pastebin.com/eAKTCt8f

mkdir build && cd build && cmake .. -DEXPERIMENTAL=1
result: https://pastebin.com/GMNa39LT

make -j 6
result: https://pastebin.com/cZjFctyJ

I see that qt version is not detected but i thought that if its files are not missing in compilation, and version is correct (I think) it would not be a problem.
qt5-base 5.15.7+kde+r173-1

I would be glad for any solution. Not necessarily for compiling, because i just want dark theme.

MichaelAtOz, 2022-11-14:
I don't believe it detects the theme, you need Edit/Preferences and set it in 3D-View & Editor.

I can't see a way to get Console, Customizer or Error-log dark.

 

  _____  

 

https://pastebin.com/q1658jmH

 

This email has been checked for viruses by AVG antivirus software.
www.avg.com

iLikeTrioxin, 2022-11-14:
@MichaelAtOz
I did get code editor and 3D-view in settings to dark mode as you said but I thought i'm missing something, because i saw that:
https://www.youtube.com/watch?v=xUZXb2FJYP8

He had opposite of my issue, dark console, frame, etc, but not editor and 3D-view.

thehans, 2022-11-14:
@iLikeTrioxin Check this FAQ entry https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#OpenSCAD_isn't_adhering_to_my_GTK_desktop_theme

iLikeTrioxin, 2022-11-14:
@thehans
Thank you very much.

closed 2022-11-14 by iLikeTrioxin