openscad/openscad #4421: segfault on render

adrianVmariano, 2022-11-21:
Runing in 2022.11.11 with all the fast-csg stuff turned on and lazy union turned on, I get a segfault and messages that read:

Error: Failed to find the next edge to walk to along the patch border!

```
include<BOSL2/std.scad>
include<BOSL2/rounding.scad>

head_diam=14.2;
shaft_diam=8;
depth=18;
thickness=2;

function slot(slotwidth, slotheight, slotradius, topradius) = let(
    angle = 85,
    slot = round_corners(
        turtle([
            "right",
            "move", slotwidth,
            "left", angle,
            "move", slotwidth/2,
            "right", angle,
            "move", slotheight,
            "left",
            "move", slotwidth,
            "left",
            "move", slotheight,
            "right", angle,
            "move", slotwidth/2,
            "left", angle,
            "move", slotwidth
        ]),
        radius = [0,0,topradius, slotradius, slotradius, topradius, 0,0], closed=false
    )
) apply(left(max(column(slot,0))/2)*fwd(min(column(slot,1))), slot);
diam = head_diam+2*thickness;
wall = thickness;
height = depth;
$fn=32;
//ycopies(n=2,spacing=26)
right_half()
    difference(){
        cyl(d=diam, rounding2=wall/2, h=height, anchor=BOTTOM);
        down(.01)cyl(d=diam-2*wall, rounding2=-wall/2, h=height+.02, anchor=BOTTOM);
        zrot_copies(n=3) //rots=[120, 240])
          bent_cutout_mask(diam/2-wall/2, wall+.03, subdivide_path(apply(back(0),slot(shaft_diam*1.2, depth, 0, shaft_diam/4)),250));
}
up(.01)cuboid([22,50,2],anchor=TOP,rounding=7,edges="Z");
```

thehans, 2022-11-21:
I was able to reproduce on Linux Mint 21 with a debug build.  Here is backtrace from gdb.  
At first I was not able to reproduce the crash (error messages were shown, but no segfault).  But that was only because I had "Stop on first warning" enabled.

Here is some of the command line output:
```
Error: Failed to find the next edge to walk to along the patch border!                         
Error: Failed to find the next edge to walk to along the patch border!                                                                                                                         
Error: Failed to find the next edge to walk to along the patch border!                                                                                                                         
Error: Failed to find the next edge to walk to along the patch border!                                                                                                                         
Error: Failed to find the next edge to walk to along the patch border!                         
Error: Failed to find the next edge to walk to along the patch border!                                                                                                                         Error: Failed to find the next edge to walk to along the patch border!                                                                                                                         
CGAL error: precondition violation!                                                                                                                                                            
Expression : CGAL::is_valid_polygon_mesh(tm)                                                                                                                                                   
File       : /usr/include/CGAL/Polygon_mesh_processing/orientation.h                                                                                                                           
Line       : 190                                                                                                                                                                               Explanation:                                                                                                                                                                                   
Refer to the bug-reporting instructions at https://www.cgal.org/bug_report.html                                                                                                                
                                                                                               
Thread 28 "QThread" received signal SIGSEGV, Segmentation fault.
```

<details>
<summary>And backtrace from GDB</summary>

```C++
#0  0x0000555555df0522 in CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Epeck, CGAL::SNC_indexed_items, bool> >::set_index (this=0x0, idx=8370)
    at /usr/include/CGAL/Nef_3/SNC_indexed_items.h:157
#1  0x0000555555ed1ffb in CGAL::Face_graph_index_adder<CGAL::SNC_indexed_items, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epeck> >, CGAL::SNC_structure<CGAL::Epeck, CGAL::SNC_indexed_items, bool>, CGAL::SM_index_pmap<CGAL::Point_3<CGAL::Epeck>, CGAL::SM_Halfedge_index> >::resolve_indexes (this=0x7ffff02bc1e0) at /usr/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h:159
#2  0x0000555555ed14f9 in CGAL::polygon_mesh_to_nef_3<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epeck> >, CGAL::SNC_structure<CGAL::Epeck, CGAL::SNC_indexed_items, bool>, CGAL::SM_index_pmap<CGAL::Point_3<CGAL::Epeck>, CGAL::SM_Face_index>, CGAL::SM_index_pmap<CGAL::Point_3<CGAL::Epeck>, CGAL::SM_Halfedge_index> > (P=..., S=..., fimap=..., himap=...)
    at /usr/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h:316
#3  0x0000555555e85585 in CGAL::polygon_mesh_to_nef_3<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epeck> >, CGAL::SNC_structure<CGAL::Epeck, CGAL::SNC_indexed_items, bool> > (sm=..., snc=...)
    at /usr/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h:337
#4  0x0000555555e85370 in CGAL::Nef_polyhedron_3<CGAL::Epeck, CGAL::SNC_indexed_items, bool>::Nef_polyhedron_3<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epeck> > > (this=0x5b4b1ade790, pm=...)
    at /usr/include/CGAL/Nef_polyhedron_3.h:627
#5  0x0000555555e85251 in __gnu_cxx::new_allocator<CGAL::Nef_polyhedron_3<CGAL::Epeck, CGAL::SNC_indexed_items, bool> >::construct<CGAL::Nef_polyhedron_3<CGAL::Epeck, CGAL::SNC_indexed_items, bool>, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epeck> >&> (this=0x7ffff02bc730, __p=0x5b4b1ade790, __args=...)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/ext/new_allocator.h:162
#6  0x0000555555e8503d in std::allocator_traits<std::allocator<CGAL::Nef_polyhedron_3<CGAL::Epeck, CGAL::SNC_indexed_items, bool> > >::construct<CGAL::Nef_polyhedron_3<CGAL::Epeck, CGAL::SNC_indexed_items, bool>, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epeck> >&> (__a=..., __p=0x5b4b1ade790, __args=...)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/alloc_traits.h:516
#7  0x0000555555e84da8 in std::_Sp_counted_ptr_inplace<CGAL::Nef_polyhedron_3<CGAL::Epeck, CGAL::SNC_indexed_items, bool>, std::allocator<CGAL::Nef_polyhedron_3<CGAL::Epeck, CGAL::SNC_indexed_items, bool> >, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epeck> >&> (this=0x5b4b1ade780, __a=..., __args=...)
[... cut ...]
#35 0x00007ffff496fa00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```
</details>

So, the behavior for CGAL errors is set such that it should throw an exception, which our code catches, thus avoiding a segfault.

From the command line output, it looks like this is happening, but then some further processing occurs which results in a crash.
Without digging into the source code yet, I am guessing that the exception handling part here is not properly cancelling the overall operation but attempting to continue working on this bad data.

kintel, 2024-11-28:
Closing as fast-csg has been deprecated

closed 2024-11-28 by kintel