openscad/openscad #4423: History breaks with projects over 25-30kb after an hour with latest Windows release.

PeppercornFox, 2022-11-25:
Unsure about other operating systems. Any project over 25-30kb - including referenced scad files with the use and include commands - can cause history to break after 45 minutes to an hour, sometimes even less. The only indication is when you use undo and it rolls back half an hours work with no way of getting it back unless backup has caught it.

This does not happen on the latest snapshot, but that has its own issues at the moment which makes working with it difficult.

I beleive this may be related to #4370

t-paul, 2022-11-25:
Are you saying the issue is fixed in the dev snapshot version?

PeppercornFox, 2022-11-26:
It seems fixed in the latest snapshot, or I haven't had it happen yet.

cjmielke, 2024-04-20:
This just happened to me, and I lost about 12 hours worth of work. Ctrl-Z randomly took me back some 50 changes, and it couldn't "redo" them. Since I "ctrl-s" so often to see the preview, the file was clobbered instantly. 

Super bummer. I just discovered OpenSCAD and love it. Looks like I better switch to a dev release.

Ubuntu 22.04, with the following packages installed via apt
`ii  openscad                                                    2021.01-4build1                         amd64        script file based graphical CAD environment
ii  openscad-mcad                                               2019.05-1                               all          library for the OpenSCAD 3D modeling software
`

I had three open files, and the program was running for several days.

MarkusNaeher, 2024-12-17:
Seems to be independent of size. Just happened to me with an 8K file.

kintel, 2024-12-17:
@MarkusNaeher Which version of OpenSCAD did you experience this with?

MarkusNaeher, 2024-12-17:
2021.01

kintel, 2024-12-17:
ok, according to the above reports, this has been fixed and working when using a development snapshot, so I'll close this as fixed since we're not planning to go back and patch old versions of the software.

closed 2024-12-17 by kintel

MarkusNaeher, 2024-12-17:
Unfortunately, there is no package of the snapshot or nightly version for my distribution, and I don't know how to build such a package myself. I also could not find a roadmap. Can you give me a raw estimation when the next official release will be there?

t-paul, 2024-12-17:
@MarkusNaeher a couple of people spending some spare time when available means not much of a timeline possible. What distro is that? There's a number of non-distro specific packages like AppImage, Flatpak or Snap.

kintel, 2024-12-17:
A new release is getting close though. I will guess 2025.XX as the next official release.