openscad/openscad #4424: Make the "Stop Render" button work as well as quitting the program completely

DeflateAwning, 2022-11-25:
Quitting OpenSCAD works better than the stop render button. I want the stop render button to quit the current render. 

I also want the "Stop Render" button to be added to the toolbar so it's super easily accessible. 

Perhaps a system-level redesign is required to spawn the renderer in a separate easy-to-quit process/thread.

MichaelAtOz, 2022-11-25:
ATM it is in a separate thread.
Unfortunately each call to CGAL is an atomic process, so it can only check when a particular CGAL operation returns.
There would be many calls in a non trivial model, unfortunately the last call is union()'ing everything and that tends to be the longest step. I suspect doing anything more drastic would have memory issues.

DeflateAwning, 2022-11-25:
Would love to look into what's required to either kill the thread forcefully, or spawn it in a separate process so as to be able to kill the process easily. 

Thoughts?

MichaelAtOz, 2022-11-25:
That's above my pay/tech grade.
If you can kill the tread without compromising the cache etc that it is both using and building that would be good.
A separate process will not be able to use existing cache, unless some shared memory is involved, then you have the same difficulty as above.
There is an incomplete (stalled) persistent cache issue here somewhere that could be involved if it is or can be transactionalised.

But this needs comment from a core-developer.

tkircher, 2022-12-05:
I'd also like to see this. Once in a while I will accidentally click render instead of preview and stop render sometimes takes ten or fifteen minutes. I obviously just quit the program instead and once in a while have to recover my recent edits from the save file. Maybe not critical but a convenience issue.

SDwarfs, 2024-10-06:
A working solution would be to clear the cache when the render is aborted. Since some bug in the code can cause the render to hang effectively forever, you could easily loose the whole code. And clearing the cache is a very small price compared to loosing unsaved code of multiple hours just because of an easy bug. Since the whole UI hangs, you cannot simply save the code and just kill the whole program without loosing the code.... and this feature would solve the issue for me.

jordanbrown0, 2024-10-06:
It's not trivial to recover, but there should be a backup automatically saved before each run.
```
Saved backup file: C:/Users/Jordan/OneDrive/Documents/OpenSCAD/backups/all-backup-itlaUEvK.scad
```

SDwarfs, 2024-10-06:
> It's not trivial to recover, but there should be a backup automatically saved before each run.
> 
> ```
> Saved backup file: C:/Users/Jordan/OneDrive/Documents/OpenSCAD/backups/all-backup-itlaUEvK.scad
> ```

You just saved my day! Many many thanks!

PS: It's C:\Users\<USERNAME>\Documents\OpenSCAD\backups on my PC ...

jordanbrown0, 2024-10-06:
Hint:  File/Show Library Folder..., then up one directory level, then down into "backups".

jordanbrown0, 2024-10-06:
But an "Open Backup" or "Recover from Backup" option might be good.

SDwarfs, 2024-10-06:
PS: the filename of the backup seems "wrong" (seems it used some filename of another open file tab; but I found it by content and file date/time). Just for the case anyone else finds this info here...

SDwarfs, 2024-10-06:
> But an "Open Backup" or "Recover from Backup" option might be good.

Crash detection with restore... meaning: every x minutes AND always before the render starts, all data from the editor somewhere (central backup, e.g. JSON-File) is stored. This file is deleted at any clean shutdown of OpenSCAD... and when OpenSCAD starts it checks if such a file exists and asks the user, if it shall restore it... however without automatically starting a render and not automatically rendering when saving (Recovery Mode). This recovery mode state could be shown in the render view with an option to return to normal. This would allow to make changes to the code to fix it and to also save unsaved files without running into the same issue again that caused the crash or hanging of the program.

DeflateAwning, 2026-05-26:
Ah yes. All issues which just sit there are therefore complete.

jordanbrown0, 2026-05-27:
I happen to agree that the stalebot is a bad idea.

However, complaining about something that other people are working on in their spare time, and giving to you for free, displays a certain misunderstanding of the relationship.