openscad/openscad #4427: list variable assign to another variable

sugizo, 2022-11-28:
**code**
```
color = ["red", "green", "blue"];

/*[ back_side ]*/
color_back_side = "red"; // color
```

**result** 
show just the red color value (no drop down menu)

**expected result**
show the drop down menu to choose the color value (red, green or blue), something like :

```
/*[ back_side ]*/
color_back_side = "red"; // ["red", "green", "blue"]
```

**question**
how to achieve it in openscad ?

thanks and best regards

UBaer21, 2022-11-29:
```
/*[ back_side ]*/
color_back_side = ["red", "green", "blue"];
```
is the solution to have the list assigned, the customizer can not read variables from comments. Also variables can not assigned in customizer as these become strings. 
For a dropdown you need to list all options as you already did.
You can convert the customizer inputs later into something else if needed.

```
color = ["red", "green", "blue"];
colorCUST=0; // [0:red,1:green,2:blue]
colorUSED=color[colorCUST];
```

sugizo, 2022-11-29:
**code in python** 
```
color = ["red", "green", "blue"]
back = color
print(back)
```

**result**
 ["red", "green", "blue"]

**suggestion**
it's more efficient, if openscad can do like that, imagine if, there is a lot of side (front, back, left, right, up, down, diagonal, etc), 
then the users must redefine the list of colors again over all of sides

**e.g. inefficient, current openscad**
```
/*[ back_side ]*/
color_back_side = "red"; // ["red", "green", "blue"]

/*[ front_side ]*/
color_front_side = "red"; // ["red", "green", "blue"]
```

**e.g. efficient, nice to have in openscad**
```
color = ["red", "green", "blue"];

/*[ back_side ]*/
color_back_side = color;

/*[ front_side ]*/
color_front_side = color;
```

UBaer21, 2022-11-29:
you can leave it open so it is just a text and you can fill every color you want.
I understand that you want this, besides copy the "comment" with the customizer definition down is not so much work. It would make changes to that list easier.
The customizer extract data from a comment - which is already not great. How should it differentiate between variables and comments then? So some list identifier would be needed. 
`a="blue";/// color`

there is already  #4291 which could be updated with general lists or the "webcolors"

dainbrump, 2024-06-09:
> ```
> /*[ back_side ]*/
> color_back_side = ["red", "green", "blue"];
> ```
>
> is the solution to have the list assigned, the customizer can not read variables from comments. Also variables can not assigned [...]

I see what @UBaer21 is trying to accomplish. The "solution" you provided of listing all options in the comment is fine if you only have a few options. But it becomes unmanageable if you have several dozen or more options you want to use.

In my use case, I have a "text color" and "background color" for my project. I want those to be a pick list for preview purposes, specifically. Both values use the same list of colors and that list of colors could reach upwards of 100 or more with new values being added to the list as needed. It would be great if there was a way to reference the list for the pick list customizer some other way without having to duplicate efforts or copy / paste big lists into comments. Also it would be ideal to keep this all contained in the scad environment instead of using an external solution like Python to touch up the scad source file which is what I currently rely on. I literally have a Python script "regenerate" my scad file with the list comments anytime I need to add another value. This introduces a whole new can of worms that could be eliminated.

I'm assuming that there is some sort of tokenizing of the comments happening otherwise how would the customizer know to make a pick list out of a list of values that are in a comment string? So maybe a `customizer` class could be triggered during the tokenization and allow something akin to the following...
```
colorlist = ["red", "black", "blue"];

textcolor = "red"; // customizer.list(colorlist)
background = "black"; // customizer.list(colorlist)

// Where 'customizer.list()' triggers the customizer to generate a list from a supplied variable.
// This would open the door to additional 'customizer' tweaks.
```

Just a thought. I'm not sure how involved this kind of "solution" would be so please keep that in mind. This is not a critique as I love OpenSCAD, but it would certainly make many things a lot easier to manage.