openscad/openscad #4430: Compilation from sources failed on linux

RileyStarlight, 2022-11-29:
Last friday I tried to compile openscad from source, following the process explained in the readme.
The process failed after a while compiling dependencies using `./uni-build-dependencies.sh`.
The error message:
```
CMake Error at cmake_install.cmake:50 (file):
  file INSTALL cannot find
  "/tmp/openscad/scripts/openscad_deps/src/CGAL-5.3/demo/resources": No such
  file or directory.
```

Machine running Linux Mint 21, 5.15.0-53-generic #59-Ubuntu x86_64 GNU/Linux

thehans, 2022-11-29:
`uni-get-dependencies.sh` is preferred over trying to build them all yourself with `uni-build-dependencies.sh`.  But neither of them tend to be maintained diligently enough to work on all systems at all times.

I am on Linux Mint 21 and rather than build CGAL myself, I use @t-paul's OBS repo:
```
deb https://download.opensuse.org/repositories/home:/t-paul:/cgal/xUbuntu_22.04/ ./
```

That repo is complementary to the development snapshot (aka nightlies) which are documented on the [openscad download page](http://openscad.org/downloads.html#snapshots) (scroll down a bit for a list of deb repos)
```
deb https://download.opensuse.org/repositories/home:/t-paul/xUbuntu_22.04/ ./
```

If you still want to build dependencies yourself and get that working, it would help to have a full output log (as a file attachment)

t-paul, 2022-11-29:
That's CGAL bug, we need to bump the version now that newer releases are available. But as @thehans said, building all dependencies from source is more meant for systems that don't have all the dependencies packaged.

t-paul, 2022-11-29:
Ref: CGAL/cgal/issues/5852