openscad/openscad #4431: unicode in labeled combo box for string

sugizo, 2022-11-30:
```
text = "\u2654"; // ["\u2654":white king, "\u2655":white queen]
text = "\u2654"; // ["\\u2654":white king, "\\u2655":white queen]
text = "\u2654"; // [\u2654:white king, \u2655:white queen]
text = "\u2654"; // [\\u2654:white king, \\u2655:white queen]
```

**result**
```
u2655
\u2655
\u2655
\\u2655
```

**objective**
show unicode in labeled combo box for string 
for example on code above shown the white king (♔) and queen (♕) when combo box is selected 

**question**
how to achieve it in openscad way ?

thanks and best regards

MichaelAtOz, 2022-11-30:
Every thing from the '//' are comments, plain text.

//suite
a=1;// [♔king,♔queen]

 

Also, for general questions/help please use the OpenSCAD Mailing-list <https://lists.openscad.org/list/discuss.lists.openscad.org> .

GitHub is used for Development and probable Bug Reports. 

If unsure of a possible bug, please ask on the Mailing-list first.

  _____  

 

 

This email has been checked for viruses by AVG antivirus software.
www.avg.com

MichaelAtOz, 2022-11-30:
Oops, test it first!

 

//suite
a=1;// [0:♔king,1:♔queen]
echo(a);

 

  _____

sugizo, 2022-11-30:
perhaps not getting what my objective is

objective is to create the drop down combo list
when user change it, let say 'white king'
then it will change the variable text, that will show the unicode white king in openscad

this work but when turn it into variable is not work
`text("\u2654", size = 10, font = "Caslon");`

**not work**
```
chess_text = "\u2654"; // ["\u2654":white king, "\u2655":white queen]

text(chess_text, size = 10, font = "Caslon");
```

perhaps openscad repo can activated the github discussion feature

UBaer21, 2022-11-30:
Seems escaping doesn't work either .. it is not converted in a symbol anymore.
```
chess_text = "\u2654"; // ["\u2654", "\\u2655"]

echo("\u2654",chess_text);
text(chess_text, size = 10, font = "Caslon");
text("\u2654", size = 10, font = "Caslon");
```

MichaelAtOz, 2022-11-30:
> perhaps openscad repo can activated the github discussion feature

NO. GitHub is for Development purposes including bug reports, and only for matters which cannot be solved by the Mailing-list. 
It is not to teach people how to code in OpenSCAD.

There are people on the Mailing-list, including many Developers, who can help you there.

Escaping is a property of STRINGS, ie part of statements or expressions.
Anything after `//` is UTF text, it is ignored by the main parser, and only text is used by customizer, not strings.
So anything special-UTF in comments just need to be pasted in as UTF characters/symbols/glyphs.

```
//suite
a=1;// [0:♔king,1:♕queen]
chesstext = (a==0) ? "\u2654" : "\u2655";
echo(a,chesstext);
```
`ECHO: 0, "♔"`

MichaelAtOz, 2022-11-30:
BTW, Customizer has external syntax dependencies with Thingiverse. 
Any changes would need to be carefully considered, so using techniques like above is currently best practice.

UBaer21, 2022-11-30:
```
textFix = 9812; // [9812:"♔", "♕"]
echo(chr(textFix),ord(chr(textFix)));

text(is_num(textFix)?chr(textFix):textFix, size = 10, font = "Caslon");
```

UBaer21, 2022-11-30:
> Anything after `//` is UTF text, it is ignored by the main parser, and only text is used by customizer, not strings.

@MichaelAtOz  But there is some parsing done by the customizer as some are converted to strings and as the escaping does produce the same echo.  Or why is the "\" removed also "" or added for text if if not there but not for numbers. I can understand that this is bit confusing. But text is also not displayed with »"« which makes [1,"1"] look equal in the dropdown box , And as far as i can see it is not documented. 
```
cust=1;//[a,1,"1",""1""]
echo(cust);
```

sugizo, 2022-12-01:
**code work**
text = "♔"; // [♔:King, ♕:Queen, ♖:Rook, ♗:Bishop, ♘:Knight, ♙:Pawn, ♚:king, ♛:queen, ♜:rook, ♝:bishop, ♞:knight, ♟:pawn]
font_family = "Caslon"; // [Caslon, Chrysanthi Unicode, DejaVu Sans, DejaVu Sans Mono, FreeSerif, MV Latin, Pecita, Quivira, Scid Chess Traveller, Symbola]

**problem** 
it's just the font family, if already read the unicode, don't bother using 'code point' just copas the 'symbol' will solve the problem

closed 2022-12-01 by sugizo