openscad/openscad #4436: Output curved surfaces

glechapelain, 2022-12-06:
I have a model that is mostly curved.

I would love it if exporting to Wavefront .OBJ or Collada would write out curves in their parameter form.

I understand it would be a major undertaking, to say the very least.  :) 
I figured I'd get the ball rolling, by throwing a rock in the pond, if you will, nonetheless.

[space_station.txt](https://github.com/openscad/openscad/files/10162739/space_station.txt)

jordanbrown0, 2023-11-29:
See also #4605, which discusses circles.

glechapelain, 2023-12-02:
Thank you so very kindly, @jordanbrown0 !  I was just about to get started on this.. on the Blender-side.
Somehow I was set-off for doing it after, but in FreeCAD, though:  My understanding is, in OpenSCAD, the tessellation occurs *prior* to the logical operation. (due to the library being used.) 

OK I just read now the other issue,  That is what I had understood.
I will close with this comment for now: and will take my business to FreeCAD (I still need OpenSCAD to `program' the model, FreeCAD doesn't really do that.  But I *think* it has the ability for the curve output (with additional changes) in addition of dealing properly the OpenSCAD format..  For completeness: this was my issue there https://forum.freecad.org/viewtopic.php?f=3&t=75373
(Well, my forum blurb.  Looks like I never ended-up opening an issue but merely commented there https://github.com/FreeCAD/FreeCAD/issues/6008#issuecomment-1380347219.  I'll keep it at it.  I am just dancing around) 

Thank you again! (That's one ball that stopped rolling :) )

closed 2023-12-02 by glechapelain

gsohler, 2023-12-02:
> merely commented there FreeCAD/FreeCAD#6008 (comment)

The ball is not really stopped ...
Once python is possible within openscad, you can import a brep kernel like
this:

https://imgur.com/a/NkvKCQ2

build123d is another approach to design solids, it uses a brep kernel.
right now i am working to get is working out-of-the-box

kee posted ...

glechapelain, 2024-01-03:
Ah good to know, thank you !  This would definitely be more convenient