openscad/openscad #4442: eliminate the need for fudge

krhackbarth, 2022-12-08:
Your tutorial says "When two objects are exactly touching each other a small overlap will be always guaranteed by subtracting or adding a tolerance of 0.001 units."

I've been doing this for all my projects and it causes an undue amount of additional coding, and can be a source of error when I forget that I should be adding the fudge factor rather than subtracting it.  

It has to be vanishingly rare that anyone would put two objects directly in contact and not intend for those objects to be "joined".

Please provide a "directive" (not sure what the proper terminology would be) that one can insert globally into the code that has the effect of telling OpenSCAD to treat all contiguous shapes as "joined".

If that means that OpenSCAD does a 0.001 mm translate behind the scenes, that's fine.  It's in the noise given the precision of my 3D-printer.

Thanks,
Ken

t-paul, 2022-12-08:
So far nobody came up with a magic strategy that just works without any side effects. The best option seems to be more special cased solutions in library components. If there's anyone with an algorithm that "just works" please post that. Sorry, until then I'll close the issue as there's no known path forward at this point.

closed 2022-12-08 by t-paul

jordanbrown0, 2022-12-09:
The problem is that "contiguous" is not well-defined.  Because of floating point inaccuracy, if you calculate the same point twice the value might not work out to be exactly the same.  (The classic example in elementary math is that 1/3 is 0.333..., and if you add 1/3 + 1/3 + 1/3 you should get 1.0 but you might get 0.999...)

*Mostly*, simple unions are not a problem and you can just place and size things at their natural size without problems.  (I have heard that there are problem cases, but I don't think I've ever encountered one.  I think they might be related to cases where the errors mentioned above cause some, but not all, of the vertexes on one face to be identical to the vertexes on another face, falling into the 2-manifold problem mentioned below.)

Differences, intersections, and unions involving multiple colors cause Z-fighting (sparkly patterns) in preview.  They are the places where you generally need to make one part slightly larger so that the surfaces are *not* identical.  (You might ask that the program identify and handle those identical surfaces; that's approximately the same problem as recognizing them above.)

There's another class of problem - the "not 2-manifold" problem, where two 3D objects precisely share *one* edge.  The easy example is two cubes:  cube(10); translate([10,0,0]) cube(10);.  It's not clear whether this "error" causes real problems, but when you encounter it it might be best to track down the offending edge and nudge it just a little.

jordanbrown0, 2022-12-09:
Edit:  Differences, intersections, and unions involving multiple colors +can+ cause Z-fighting