openscad/openscad #4450: s = join([s1, s2, ...], sep)  - concatenate an array of strings, optionally with a separator

jordanbrown0, 2022-12-18:
It's annoyingly difficult to assemble a string with variable contents.  If you have an array of values, and you want to assemble them into a string somehow, you have to write a recursive function.  You can generate an array using list comprehension, but there's no equivalent for generating a string.

It would be nice if you could have something akin to JavaScript's Array.join() and Python's str.join().

I suggest
```
s = join([s1, s2, ...]);
s = join([s1, s2, ...], sep);
```

Without the "sep" argument, concatenates the strings from the array.
With the "sep" argument, concatenates them separated by that string.

You can implement this straightforwardly using recursion, but it seems useful enough that it would be nice for it to be ubiquitous.  Maybe we should get some library to be ubiquitous, and then define this function in it.

MichaelAtOz, 2022-12-18:
Should `concat()` get enhanced perhaps, rather than a plethora of join mechanisms?

jordanbrown0, 2022-12-19:
Maybe, though I'm not sure exactly how you would differentiate "concatenate the strings in this array" from the existing cases.

And note that both JavaScript and Python call this operation "join".  (Albeit they are backwards from one another; in JavaScript it's a method on an array while in Python it's a method on a string.)

thehans, 2022-12-19:
My funcutils library has a fancy recursive join that uses concept of binary divide and conquer, and is quite fast even for huge lists

https://github.com/thehans/funcutils/blob/ab11ed7b4270de9007669bfb51a499af8f9efbf8/string.scad#L3-L9

MichaelAtOz, 2022-12-19:
I have wanted all sorts of string handling routines in OpenSCAD, that exist in most other languages.
As OpenSCAD was initially only a geometry domain specific language, there was little need. 

So like many people I built a range of recursive routines to handle the usual suspects.
Then text() chr() ord() etc extended the usability of strings. 
So the set of typical string functions is needed.
First option, start with `join()`, then progress through the [other suspects](https://en.wikipedia.org/wiki/Comparison_of_programming_languages_(string_functions))

But as I was casting my thought about, I looked at `list comprehensions`.
It would be so easy to 'process' characters in a similar way to lists.
You can change strings into lists of characters, in a number of ways, but
what is missing is changing a list of chars into a string.

I looked at `chr()`, it can take numbers (code-points), list-of-numbers, range-of-numbers.
Currently:
> chr(Vector)
>     Convert all code points given in the argument vector to a string.

```
echo(chr(65), chr(97));      // ECHO: "A", "a"
echo(chr(65, 97));           // ECHO: "Aa"
echo(chr([66, 98]));         // ECHO: "Bb"
echo(chr([97 : 2 : 102]));   // ECHO: "ace"
```
What if it could take strings, and list-of-strings?
`chr(65, " bird", [ " in", " the"], " hand");

With some fine tuning of some other things, that could make something like `string comprehensions`.

??

jordanbrown0, 2022-12-19:
Extending chr() is an interesting possibility, especially for multiple numbers or arrays of numbers.

But I have to think that the ubiquity of the word "join" for this particular function is compelling.

But but:  introducing objects introduces significant namespace opportunities, so maybe we should think about whether we want to take advantage of them.  Maybe we should grab the word "string" for a predefined object that holds string operations, and have string.join(), string.split(), et cetera, instead of grabbing a ton of individual words.

MichaelAtOz, 2022-12-20:
If it went that way, I don't think you need 'string' as a key-word. Surely you use the identifier `my_list = my_message.split()`

Note that what you are after here is a type conversion from vector. 
Your `s = join([s1, s2, ...]);` is not joining per se. 
As strings are in-effect vectors of characters, you are conceptually flattening the vector.
The 'joining' is a side effect.
I don't think that such flattening is applicable to vectors of non-characters.

`chr()` would seem suitable.

(that is disregarding the 'separator' aspect for now. You can introduce a "\<separator>" in your string vector easily with comprehensions. Alternatively add `separator=","` to `chr()` and/or `concat()`)

Don't lose the `string comprehensions` concept.
Once you have a string easily accessible as a list of characters, you can readily do a whole lot of manipulation with comprehensions.

So extend the language with a large number of string methods, or provide a mechanism to manipulate strings easily.
If there was a MACRO facility, you could have a std library with e.g. `SPLIT()`, which substitutes the appropriate comprehension.
(but need to consider performance, vs natively coded) 
The core/non-core question... & Beginner friendly, eg substring(), or power user feature...hmmmm...

My only issue is string.methods will always have people wanting another one `.to_snake_case()`...

But, please Nooooo regex

MichaelAtOz, 2022-12-20:
Or `list_to_string()`, `to_string()` etc.

& just to inflate the topic, if you go methods, we really need a comprehensive `.format()` prettyfier.

jordanbrown0, 2022-12-20:
Indeed, if we had methods on things other than the things we're currently calling objects, then the need for namespace would be even further reduced.  And maybe we should figure out how to go that way.  Getting away from the "one namespace for all functions" notion seems desirable, one way or another.

Without doing that, as for the name... I'd say that the fact that Python, JavaScript, Java, and Visual Basic all call this operation "join" is a pretty compelling reason for choosing that name.

We already have iteration across a string.  Combining that, list comprehensions, and a "join" operation gives you string comprehensions, I think.

We could have a discussion on REs, if anybody ever thought that we needed that kind of functionality, but let's not do it here.

jordanbrown0, 2022-12-20:
Filed #4455 for a place to discuss the general `val.method()` concept.  This join function should wait for a resolution there - that's a better way to structure the namespace.

MichaelAtOz, 2022-12-20:
> Python, JavaScript, Java, and Visual Basic all call this operation "join"

Well, OCaml uses `concat()` ... ;)

Wiki
```
str
Convert all arguments to strings and concatenate. 
```

If you want to leave `join()`, and the plethora of methods, until the wider support is there.
`str(<args>, separator="<sep_arg>")` could provide the capability, in old-style OpenSCAD now.
```
echo( str( [ "ab", "cd" ], separator="") ); // -> "abcd"
echo( str( [ "ab", "cd" ] ) ); // -> "["ab", "cd"]" (as it does now, convert vector to echo() equivalent text)
echo( str( [ "ab", "cd" ], separator=",") ); // -> "ab,cd"
// decision, does 'separator=' only apply to vectors (of any type, or only string), or also to multiple args to str()
// eg str( separator=".", "three", 1415926) -> "three.1415926"
```
I'd say `separator=` makes `str()` use the separator between all non-vectors args,
and vector with top level elements of any type, so top level strings are joined with the separator,
but if a top level element is a vector, ie nested, that is output as echo writes a vector.
`str( [ "ab", "cd", [ "cat", 1 ], ["dog", 2] ], 42, separator="|" ); //-> "ab|cd|["cat", 1]|["dog", 2]|42"`

Because currently
```
vp=["pick", "a", "peck", "of"];
s=str(vp, "pickled", "peckers");
echo(s=s);
```
`ECHO: s = "["pick", "a", "peck", "of"]pickledpeckers"`

MichaelAtOz, 2022-12-20:
...and if you want to get fancy, multiple `separator=`, in which case the args following use the specified separator.
This would allow some fancy formatting.

swistak, 2025-08-01:
If someone stumbles on this ticket here's my take on this that works in stable version:

```
// Joins strings with an optional separator and limit
// 
// join(["A", "B", "C"]) -> "ABC"
// join(["A", "B", "C"], ",") -> "A,B,C"
// join(["A", "B", "C"], ",", 2) -> "A,B"
//
function join(parts, sep="", limit = undef) =
  let(n = is_undef(limit) ? len(parts) : limit)
    n == 0 ? "" :
    n == 1 ? parts[n-1] : 
    str(join(parts, sep, n-1), sep, parts[n-1]);
```